sandeepmistry/arduino-LoRa #229: Receiving always the same data

Mantisworks, 2019-02-11:
Good Afternoon,
    I'm approaching LoRa on Arduino board. Actually I'm reading Temperatura and Humidity from a DHT22 sensor and sending them to another LoRa board... The thing is that monitoring the TX serial data and breathing near the sensor all datas changes... But the Receiver board read the first packet e never acquire new datas from TX board. Datas changes only if i reboot the receiver..

Did you ever had this kind of issue?

Thank you!!!!!

Mantisworks, 2019-02-12:
Solved!

Just commented the line n. 484 in LoRa.cpp

`//writeRegister(REG_FIFO_ADDR_PTR, 0); // Commented by me`

Regards
Ruben

closed 2019-02-12 by Mantisworks

mlhfb, 2019-03-06:
Thanks I pulled my hair out for days expecting it was my n00b code...  It ended up being ln 583 for me....  My question is... Is this a bug or a feature?  Is there another workaround that doesn't involve modifying the library?  Thanks!!!