sandeepmistry/arduino-LoRa #230: Range Issues with LoRa

deepakjayan93, 2019-02-15:
I have mainly 2 doubts,
1) I connected my LoRa transmitter (SX1276 OF 868Mhz) to arduino NANO 3.3 V pin and LoRa receiver to arduino UNO (3.3V pin). The SPI frequency was set to 8MHz. The spreading factor was initially 7 and later, I changed it to 12. The signal bandwidth was set to 125kHz. I checked the whole thing inside my building . The transmitter was kept in the second floor and I carried the receiver LoRa to ground floor. I received packets only within a range of 100 metres. In the ground floor in some places, I am losing the packets. what could be the reason for this ? Actually I need to get a range close to 2000km, as per SX1276 datasheet for a tx power of 17 -20 dBm and a spreading factor of 12?
2)// subtract 3 from level, so 18 - 20 maps to 15 - 17
      level -= 3;
I saw the above line in LoRa.cpp code.
[image]
 Why do we subtract 3 from Tx power? Please reply.

lodesmets, 2019-02-19:
I found this in the datasheet
[image]
I think it has something to do with this formula used. This register is set in this line writeRegister(REG_PA_CONFIG, PA_BOOST | (level - 2));
No idea why, but it is needed, otherwise you can get negative Pout

sobehard, 2019-03-02:
Hi, @deepakjayan93. As mention in the ReadMe document, direct connect the source (3.3 V) from the Arduino Nano to the LoRa module may not provide sufficient current for work properly. Maybe you can try to connect an external source (battery) to the LoRa module and see whether it can solve your problem or not. (*insufficient current will affect the actual transmit power, hence, the receiver may not able to detect the low power signal from the sender as distance move further)