sandeepmistry/arduino-LoRa #231: ESP8266 with arduino-Lora library?

panosss, 2019-02-24:
Hi everybody.
I have made a receiver and a transmitter with 2 arduinos and 2 Lora RA-01 modules using your library, arduino-Lora.
I replaced on the receiver the Arduino with an ESP8266-12E module.
But it's not working anymore.
Can this library work with ESP8266?
Should I make any changes?
Thank you.

closed 2019-02-24 by panosss

JRAlvarez2017, 2019-11-12:
Could you tell me the pins wiring? you used I cannot make work with an ESP8266
Thanks

panosss, 2019-11-12:
NodeMCU.........LORA
GPIO15(D8)......SS
GPIO13(D7)......MOSI
GPIO12(D6)......MISO
GPIO14(D5).....SCK
GPIO0(D3).......RESET

```
const int csPin = 15;         // LoRa radio chip select                                    GPIO15(D8)->ss OF Lora module
const int resetPin = 0;       // LoRa radio reset                                          GPIO0(D3)->RESET OF Lora module

LoRa.setPins(csPin, resetPin);
```

Erik84750, 2020-11-24:
I have the same issue: lora.h works with Arduino, not with ESP8266 (MISO: GPIO12; MOSI: GPIO13; CLK: GPIO14; CS: GPIO15): are the pin definitions in the library correct? Where to connect and define RST and NSS (chip select)?

Erik84750, 2020-11-24:
From: [https://github.com/sandeepmistry/arduino-LoRa](url)

_NSS, NRESET, and DIO0 pins can be changed by using LoRa.setPins(ss, reset, dio0). DIO0 pin is optional, it is only needed for receive callback mode. If DIO0 pin is used, it must be interrupt capable via attachInterrupt(...)._

Is that the solution?