sandeepmistry/arduino-LoRa #232: Numbers don't show correctly

panosss, 2019-02-24:
My Sender sends a message with this code:

> void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);
  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();
  counter++;
  delay(5000);
}

My Receiver receives with this code:
> void loop() {
  display.clearDisplay();
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    display.setTextSize(1);
    display.setTextColor(BLACK);
    display.setCursor(0,0);
    display.println("Received packet '");
    display.display();
    // read packet
    while (LoRa.available()) {
      display.print((char)LoRa.read());      
      display.display();
    }
    // print RSSI of packet
    display.print("' with RSSI ");    
    display.println(LoRa.packetRssi());
    display.display();
  }
}

The result on the receiver 's screen (5110 with the library Adafruit_PCD8544):
Received packet 'el 1 ' with RSSI  -24   (instead of 10, or 11 or 12 etc it displays 'el 1 ')
or: Received packet 'el 2 ' with RSSI  -24 (instead of 20, or 21 or 22 etc it displays 'el 2 ')

Maybe somebody knows how to fix this?

YogoGit, 2019-02-24:
```
// read packet
while (LoRa.available()) {
display.print((char)LoRa.read());
display.display();
}
```
Try:
```
// read packet
while (LoRa.available()) {
  display.print((char)LoRa.read());
}
display.display();
```

panosss, 2019-02-25:
I tried it, no change.

panosss, 2019-02-26:
Solution: in the tsansmitter I had to convert the int counter to String:
`// send packet
  LoRa.beginPacket();    
  LoRa.print(String(counter));
  LoRa.endPacket();  `

closed 2019-02-26 by panosss