sandeepmistry/arduino-LoRa #236: when i use lora.endpacket(true); cpu halting

halukmy, 2019-03-18:
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40090705  PS      : 0x00060a34  A0      : 0x80091899  A1      : 0x3ffb54b0  
A2      : 0x3ffbdd50  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060823  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0x3ffb5640  
A10     : 0x00000000  A11     : 0x3ffc5aac  A12     : 0x00000009  A13     : 0x0000000c  
A14     : 0x00000016  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x40090705:0x3ffb54b0 0x40091896:0x3ffb54e0 0x4008fc87:0x3ffb5500 0x4008fd91:0x3ffb5540 0x4008469d:0x3ffb5560 0x40085159:0x3ffb5580 0x400888e9:0x3ffb5620 0x4008700f:0x3ffb5640 0x4008766d:0x3ffb5670 0x4012a45a:0x3ffb56b0 0x4012a5aa:0x3ffb56e0 0x40089015:0x3ffb5710 0x4008f00d:0x3ffb5750

Core 1 register dump:
PC      : 0x40092112  PS      : 0x00060d34  A0      : 0x80091272  A1      : 0x3ffbe800  
A2      : 0x3ffb8460  A3      : 0x3ffb8074  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060d23  A7      : 0x00000000  A8      : 0x3ffb8074  A9      : 0x3ffb8074  
A10     : 0x00000018  A11     : 0x00000018  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060d21  A15     : 0x00000000  SAR     : 0x0000000e  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x40092112:0x3ffbe800 0x4009126f:0x3ffbe820 0x4008fd27:0x3ffbe840 0x400d72dd:0x3ffbe880 0x400d45fd:0x3ffbe8a0 0x400d4730:0x3ffbe8d0 0x400d4764:0x3ffbe8f0 0x400d4d9d:0x3ffbe910 0x400d4e06:0x3ffbe930 0x40080f8d:0x3ffbe950 0x40081e31:0x3ffbe970 0x4000bfed:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4
LoRa init succeeded.

LoRa Simple Gateway
Only receive messages from nodes
Tx: invertIQ enable
Rx: invertIQ disable

if i use    LoRa.endPacket(true );                 // finish packet and send it

if i not use (true),, just so slowly works..

why is it?

halukmy, 2019-03-22:
i want share more detail,

when i up spread spectrum to 12, this problem begins..

spread spectrum 12 and lora.endpacket(true) = boooom...

halukmy, 2019-03-22:
but if not use endpacket or only lora.endpacket(); no problem but, continus messages are terrible slows

universam1, 2019-03-22:
Do you have a yield statement in your sketch? Are you aware that the esp needs a watchdog reset within a second

morganrallen, 2019-06-04:
This issue is related to what @universam1 is suggested, if you hold the main loop the ESP will trigger the Watch Dog Timer and reset

closed 2019-06-04 by morganrallen