sandeepmistry/arduino-LoRa #237: Wrong RSSI values in the range 779 - 868 MHz

ea4eoz, 2019-03-18:
I noticed a little bug in the packet_rssi() function.

return (readRegister(REG_PKT_RSSI_VALUE) - (_frequency < **868E6** ? 164 : 157));

Frequencies between 779 and 1020 MHz should use the value 157 but currently 157 is only used in the range 868 - 1020 MHz, giving wrong rssi values for the 779 - 868 MHz range.

I would compare to a frequency midway between the end of the Low Frecuency Port (525 MHz) and the start of the High Frequency Port (779 MHz), for example 652E6. This gives correct RSSI readings in every frequency.

Best regards.

halukmy, 2019-03-29:
@ea4eoz  details please

ea4eoz, 2019-03-31:
To test the problem just pair two devices and check RSSI strength on two different frecuencies, one above and the other below 868 MHz, for example 867.900 and 868.100 MHz. You will see the 7 dB difference between both frequencies.

Take a look at section 5.5.5 (RSSI and SNR in LoRa Mode) in SX1276 datasheet taking into account the high frequency port works from 779 to 1020 MHz, as you can see on section 5.3.3 (PLL).

Thank you

halukmy, 2019-04-23:
yes this arduino library has really big problem with that..

when iset  999 on code, i get  768 mhz
or 1200  i get  778....

hope someone fix that

sergio303, 2019-09-18:
yes, mid band threshold used to differentiate high and low freq ports is wrong, I've created a PR to fix that.

https://github.com/sandeepmistry/arduino-LoRa/pull/288

also, taking a look at Semtech documentation, looks like RSSI should take into account SNR for correct calculation, I'll try to make an updated PR with this...

morganrallen, 2019-09-18:
Thanks @sergio303 I'm going to have a look at this tonight.