sandeepmistry/arduino-LoRa #240: LoRa.begin() failed

7n7n7n, 2019-04-10:
Hello guys.
First of all, please understand that i'm poor at english. because i don't live in english-speaking countries.

Problem : 
i'm using two "mkr wan 1300" for trying peer to peer communication.
so i installed this libraury. ( for LoRaReceiver, LoRaSender Example )

but i couldn't get the result i wanted.
only get the message : Starting LoRa failed!

What i did to solve :
- FW Update (1.1.9) with MKRWANFWUpdate_standalone example. also run serial monitor. 
...

Suspected :
readRegister method in LoRa.begin( ) always return 0, not 0x12.
- maybe is it hardware problem?

and i heard the USB disconnect sound while uploading the source code. 
but the upload is completed with no failure and the connection sound is heard again. 
- is it normal?

i only had a USB connection with my laptop. i did not configure the circuit at all.
- do i have to configure it? if yes, please give me some example...

Thanks

chizran, 2019-04-10:
I have the same problem using two mkr wan 1300  with the latest firmware, even after returning to the previous firmware version, I get the same error. The funny thing is it worked with the previous version of the firmware.

7n7n7n, 2019-04-11:
chizran//
I'm sorry but I couldn't understand what you said..
Do you mean you solved the problem with previous version of the firmware??
then, which version you installed?

chizran, 2019-04-11:
No I didn't manage to solve the problem. It worked before with the previous version of the firmware, but even if I flash again with the same version, it doesn't work anymore.

7n7n7n, 2019-04-11:
chizran//
oh.. sorry about that.,  thank you anyway
I hope to solve our problem quickly..

chizran, 2019-04-15:
Any update on this? Does anyone else have the same problem?

MiGAR19, 2019-04-15:
Hi!
I have two MKRWAN1300 boards with the same issue. I have tried with FW 1.16 and FW 1.19 and Library version 0.5.0 without success.

Michael-List, 2019-04-15:
There are already related issues to this in other projects.
- https://github.com/arduino-libraries/MKRWAN/issues/23
- https://github.com/arduino/ArduinoCore-samd/pull/400

MiGAR19, 2019-04-18:
Thanks. It worked out with the other Board library

closed 2019-11-29 by 7n7n7n