sandeepmistry/arduino-LoRa #247: Error on API.md

gioreva, 2019-05-06:
Supported values are 7.8E3, 10.4E3, 15.6E3, 20.8E3, 31.25E3, 41.7E3, 62.5E3, 125E3, and 250E3.

It support also 500€3

halukmy, 2019-05-06:
@gioreva  hows performances at 500e3

gioreva, 2019-05-07:
Use the calculator
https://github.com/tanupoo/lorawan_toa

And chapter 4 of the datasheet
https://www.semtech.com/uploads/documents/DS_SX1276-7-8-9_W_APP_V6.pdf

halukmy, 2019-05-07:
@gioreva  thanks, but this library got some problems, i just used machine to radio waves, and 1024 mhz nearly 700 mhz ... strange..

and another question

what is biggest distance?

sp 12 and 500e3?

gioreva, 2019-05-07:
433Mhz, in the middle of the houses and plants I did about 5km
10Cm antenna

I didn't understand your problems, I didn't have any.

halukmy, 2019-05-07:
the problem is, when we set frequency of 433e6 like something, actually its not give real 433 mhz you should check it..

sooo , can you share your antenna please?

gioreva, 2019-05-07:
Yes, is 433
But you can only change frequency at init, not after.

  if (!LoRa.begin((long)433050000+((long)Frequenza*25000))) {      // initialize ratio at 433 MHz

gioreva, 2019-05-07:
This antenna:   [link](https://www.aliexpress.com/item/RF-433Mhz-Antenna-Omni-2-15dBi-With-SMA-Male-Connector-Radio-Antenna-Size-L-165mm/32599569296.html?ws_ab_test=searchweb0_0%252Csearchweb201602_9_10065_10068_319_317_10696_10084_453_10083_454_10618_10304_10307_10820_10821_537_10302_536_10843_10059_10884_10887_321_322_10103%252Csearchweb201603_52%252CppcSwitch_0&algo_expid=64e6addd-7e93-40e6-abfc-1e62dd56c8a4-10&algo_pvid=64e6addd-7e93-40e6-abfc-1e62dd56c8a4&transAbTest=ae803_3&dp=d60ecec7215f34d0b6310592a31a68c3&af=886947&cv=47843&afref=https%253A%252F%252Fwww.aliexpress.com%252Faf%252F433-antenna.html%253FSearchText%253D433%252Bantenna%2526d%253Dy%2526initiative_id%253DSB_20190507000021%2526origin%253Dn%2526catId%253D0%2526isViewCP%253Dy%2526jump%253Dafs&mall_affr=pr3&dp=d60ecec7215f34d0b6310592a31a68c3&af=886947&cv=47843&afref=https%253A%252F%252Fwww.aliexpress.com%252Faf%252F433-antenna.html%253FSearchText%253D433%252Bantenna%2526d%253Dy%2526initiative_id%253DSB_20190507000021%2526origin%253Dn%2526catId%253D0%2526isViewCP%253Dy%2526jump%253Dafs&mall_affr=pr3&aff_platform=aaf&cpt=1557216034883&sk=VnYZvQVf&aff_trace_key=3872ae97da3247cc9912dc665d344bc9-1557216034883-05029-VnYZvQVf&terminal_id=57712335f6e943148baf84e3ad37f5ad)

IoTThinks, 2020-11-12:
Already updated with 500Khz
https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#signal-bandwidth

closed 2020-11-12 by IoTThinks