sandeepmistry/arduino-LoRa #248: Lora has failed to initialize

sumitkanu, 2019-05-09:
We are working with a LoRa/GPS hat and Arduino. Our connections are according to the connection given in the readme file and the pins of the hat are according to the ones given in its official site. Yet lora initialization fails everytime. We are using the LoRa sender code from the examples. Link to lora hat site:[LoRa/GPS hat](https://wiki.dragino.com/index.php?title=Lora/GPS_HAT). Also will the Hat that is designed for Rpi work with arduino?

wero1414, 2019-05-23:
Is checking the chip version over the SPI, the reasons for this kind of mistake usuall is the SPI please check well and double check your SPI connections, MISO MOSI SCK and SS cause if youre not even getting response of the function init, it means that your SPI communication is down

closed 2019-05-23 by wero1414

sumitkanu, 2019-05-23:
We have checked the connections multiple times. Is there some minimum current required by the hat that the arduino uno is unable to supply. And will the code work if I comment out the chip version check part from the .cpp file?