sandeepmistry/arduino-LoRa #250: SX1278 range issue

d100cent, 2019-05-19:
I'm using SX1278 433MHz module and not getting enough range with it. Kindly suggest best configuration with good antenna.

wero1414, 2019-05-23:
You should print the RSSI when receiving packets to see how much are you losing, and a guy experienced bad frequency configuration while using the SX1278 i think, so maybe theres a problem with the oscillator and then all the frontend of the RF is working bad, so if you have something like an SDR would be helpful to look at what frequency are you trasmitting

d100cent, 2019-05-23:
> You should print the RSSI when receiving packets to see how much are you losing, and a guy experienced bad frequency configuration while using the SX1278 i think, so maybe theres a problem with the [...]

I'm already printing RSSI value, issue is bad antenna received with SX1278.

I'm not able to decide which antenna to be used for at-least 2Km range.

If you can suggest, it will be very much helpful for me.

closed 2021-07-04 by IoTThinks