sandeepmistry/arduino-LoRa #253: Support for Gateway

manvendra88, 2019-06-03:
Hi! Thanks for such a great library to learn and tinker lora.
I have been using LoRaSender and LoRaReceiver code for two Dragino Lora Sheilds. So far, i am able collect the meta data for my Sender at receiver. My questions are :-
1. Can i use the same LoRaReceiver code for the radio of my LG-01 Gateway?
2. Can the LoRaReceiver code can recept from two different Senders?

morganrallen, 2019-06-04:
1) Unfamiliar with that device so it's hard to say, I would say most likely, but you'd have to implement what ever protocol they use.

2) Yes, so long as all their settings are the same.

manvendra88, 2019-06-06:
Thanks @morganrallen for the quicky reply. As i was digging to find the solution the problems, here are some useful things that i found, in case if it will help someone else too. 

1. LG-01 gateway is from Dragino and it can be easily programmed to be used by Sandeep's library (Which is great!). Dragino has a an example code that supports LG-01 (https://github.com/dragino/Arduino-Profile-Examples/blob/master/libraries/Dragino/examples/LoRa/LoRa_Receive/LoRa_Receive.ino#L3).
Moreover, if you were to use another library for LG-01, then check for https://github.com/dragino/Arduino-Profile-Examples/tree/master/libraries/Dragino/examples/LoRa.
2. I am a little confused to how to set the same settings. If I set the SF same for receiver same as that of node-1. Should i change the SF of the receiver corresponding to node-2 in a loop?

Thanks :)

manvendra88, 2019-06-06:
As the gateway issue is solved already. I shall close this issue. :)

closed 2019-06-06 by manvendra88