sandeepmistry/arduino-LoRa #256: Plans to support SX1280 2.4ghz LoRa module?

TimLeijten, 2019-06-07:
Hi,

I found this little module on ebay:
https://www.ebay.com/itm/2-4GHz-Lora-Module-Bluetooth-SX1280-E28-2G4M12S-2-4G-BLE-Wireless-RF-Transceiver/192394891752
It is a 2.4ghz LoRa transceiver powered by the SX1280 chip.
I would like to use one of those for my project because I need a bit more speed than a 900mhz LoRa module gets me, and the SX1280 chip can handle speeds up to 2Mb. Which is perfect for me.

Unfortunately, I have been unsucsessfull in finding a driver for it that works with arduino(based) boards.

So I'm wondering if anyone here is interested in developing/porting a driver for it? The SX1280 site seems to have a driver for mbed arm boards, so that would be a good start on porting it.

Is anyone interested?

morganrallen, 2019-06-07:
Neat chip but at a glance it's going to be outside the scope of this driver, at least in the near term.

closed 2019-06-07 by morganrallen

TimLeijten, 2019-06-07:
Okay, thanks.