sandeepmistry/arduino-LoRa #260: Semtech erratum: 2.3 Receiver Spurious Reception of a LoRa Signal

danak6jq, 2019-06-21:
Is there a plan to implement this erratum?

[image]

deanpretorius, 2019-07-30:
Hi I am having this same issue. I am trying to use the Dragino Shield v1.4, which has the sx1276 chip on it, and the Arduino MKR WAN 1300. Sending messages to the Dragino shield does weird things. Were you able to fix this problem?

danak6jq, 2019-07-30:
> Hi I am having this same issue. I am trying to use the Dragino Shield v1.4, which has the sx1276 chip on it, and the Arduino MKR WAN 1300. Sending messages to the Dragino shield does weird things. Were you able to fix this problem?
> 
> I am sending data from the MKRWAN at Datarate 5, SF7 BW125, at 868.1MHz. But I am still receiving these incorrect messages on the Dragino.

I don't believe this issue would result in "weird things" if you have only one LoRa transmitter; this issue results in receiving LoRa signals at another frequency. I suppose it's possible you're hearing LoRaWAN traffic in your area if you're using the LoRaWAN sync word of 0x34, but that's not the default, so this is doubtful.

deanpretorius, 2019-07-30:
I think I might be hearing it on the wrong frequency. But I am not sure how to implement this on the Dragino LoRa HAT and Raspberry Pi. As far as I am aware, my device is the only active one in the area, I live in a small student town. So it wouldn't be passing LoRaWAN traffic.