sandeepmistry/arduino-LoRa #261: CRC not working

mchacher, 2019-06-23:
Hi all,
I have activated CRC on both side (sender and receiver). My test setup was pretty simple with one receiver, and 2 senders. Since the nodes sending LoRa message can do it at the same time it is quite easy to get corrupted messages on receiver side.
Even though, the bad packets are still processed in my callback method to process the incoming data.
I did check the value of the IRQ Flags used line 642, and it remains ok ...
So something wrong but I didn't figure out how to fix it.
If someone is ready to help ...
Thanks.

JopWerff, 2019-09-27:
Hello Sandeep or anybody else,

Can some one tell me what is the sense of this enableCRC. I tried this (enabling at both sides), but still receive sometimes rubish packets. Should I check by code some CRC-OK-status or something else?
I don't see in the library functions to do so. But I'm a newbie in this area.
I would be glad if somebody will shine a light about CRC. It would be a nice feature, if it really works.

Regards,

   Jop

mchacher, 2019-09-27:
Hi @JopWerff - this is consistent with my message above. 
@sandeepmistry - any opinion on this?

sergio303, 2019-09-27:
Are you using Implicit or Explicit header mode? CRC is done on hardware on the chip (if activated), but it's up to the library to check the CRC ok flag on reception. It is very unlikely that you can receive corrupted data with a correct CRC flag and setup, so something else is going on here, I'll take a look at how the library handles it on reception.
As far as how the chip handles it, depending on mode:

[image]

JopWerff, 2019-09-27:
Hi Sergio,

Thanx for your rather extensive answer and info how the chips is handling this. It makes me also a bit dizzy. I'm rather a newbie in this area. Reading registers not yet in my skills.

As far as I can see the library doesn't have clear routines or status vars to check status of CRC validity for last received message. I like too see a library as a "black box" - not knowing how he does it - as long as he does it, so only interested in the outcome or reliable results.
It would be nice if this nice library will be extended with such stuff. In my imagination something like a switch "don't exposure a payload to outside if CRC is bad / don't do a call back on interrupt" to set simple on 'true' by stupid/lazy developers as me ;).

Let's hope for the future, this will come.

Beste regards,

    Jop

ps  Don't see my remark as criticism; I'm very glad with this library.

JopWerff, 2019-10-06:
In following-up of my earlier posts.

This is a LoRa-packet as it ought to be  in my configuration of one LoRa-sender and 2 receivers:

23:50:06.635 -> Received LoRa-packet '362/2019-10-05T23:50:00Z\20.94&182#100'
23:50:06.635 ->   --incomingPacketSize= 38  --RSSI -53

(the  time-date stamps  at left come from my Arduino IDE serial monitor)

And sometimes I got these kind of LoRa-packets:

23:55:21.571 -> Received LoRa-packet ',s⸮ڮQфO⸮⸮_Uv?⸮⸮⸮)⸮' ⸮+⸮9⸮⸮⸮
23:55:21.571 -> CW2(⸮]⸮6v2⸮Y⸮⸮i(w⸮)u⸮K⸮S⸮e⸮K⸮⸮N6⸮0C({Yl⸮⸮⸮⸮⸮C/⸮⸮K[0⸮⸮⸮:;?⸮_'
23:55:21.571 ->   --incomingPacketSize= 108  --RSSI -107

It would be nice if we could filter these out in a simple way by making use of CRC as apparently is built-in in the Semtech-chip.

Regards,

  Jop

sergio303, 2019-10-07:
Hi,

library is already checking if the CRC flag calculated by the chip on reception is ok (if it is enabled on the sender node with enableCrc() method). It should not pass a packet whose CRC is not ok if it is enabled.

Are you sure you're not receiving valid LoRa packets from a nearby node (LoRaWAN or other node in the neighbourhood using same settings as you)?  Maybe you can try to slightly change frequencies and/or SF and BW and see if it goes away (btw, what settings are you using?)

JopWerff, 2019-10-17:
Hi Sergio,

Thanx for your clarifying answer. Indeed I looks as you are right: the weird packages my receiver is receiving have probably also a correct CRC but are meant for another communication line.
Since I enabled my CRC on both sides in my communication line I didn't see packages meant for my communication with some weird characters (such as �⸮) in it. They are all correct (but dit not yet test this over great distances sender => receiver).
It is rather easy for me to let ignore these other packages as their format is totally different. So no problem.

I think my settings are rather simple - I just build a project according a Random Nerd Tutorials [example](https://randomnerdtutorials.com/esp32-lora-rfm95-transceiver-arduino-ide/#more-63668) (not exactly this one but the one in there course on ESP32). But I added an extra as more sensors and an RTC at transmitter side and made it more robust to tackle long lasting drops of WiFi; besides that I upload data to a server database see also RNT-examples.

Here my init-LoRa code (receiver side):

bool InitLoRa() {  
  Serial.print("LoRa initiëren: ");
  int count = 0;
  // Initialize LoRa
  SPI.begin(SCK, MISO, MOSI, SS); 
  
  LoRa.setPins(SS, RST, DI0);
  pinMode(DI0, INPUT_PULLDOWN);  //???   see: https://techtutorialsx.com/2017/09/30/esp32-arduino-external-interr
  //replace the LoRa.begin(---E-) argument with your location's frequency 
  //note: the frequency should match the sender's frequency
  //433E6 for Asia
  //866E6 for Europe
  //915E6 for North America
  while (!LoRa.begin(LORA_BAND * 1E6) && (count <= 20)) { 
    count ++;                                             
    Serial.print(".");
    delay(500);   
  }
  
  if (!LoRa.begin(LORA_BAND * 1E6)) {  
    Serial.println("Starting LoRa failed!");
    return false;
  }

  // Change sync word (0xF3) to match the sender
  // The sync word assures you don't get LoRa messages from other LoRa transceivers
  // ranges from 0-0xFF
  LoRa.setSyncWord(0xF3);         
  Serial.println("\nLoRa-receiver initialized OK!");

  //Activate  CRC  
  LoRa.enableCrc();

  #ifdef _InterruptDriven
  // register the receive callback
  LoRa.onReceive(onReceive);
  #endif

  // Set the radio into receive mode
  LoRa.receive();
  delay(50);
  return true;  
}

Code at transmitter side about the same, except for omitting the last lines (from #ifdef _InterruptDriven)
Should I change my SyncWord in a more exceptional one? But what is a less used value?

Did not yet bother about frequency and SF or BW. Also I have to see how I can easily extend some parameters to make the distance between sender & receiver much bigger. 
If you have good suggestions, don't hesitate. I'm rather a newbie.  Shortly ago Andreas Spiess has published a video on that subject. Hadn't time to see this seriously.

Thnx Sergio and best regards,

    Jop

sergio303, 2019-10-28:
As for SyncWord values, 0x12 is the default after chip reset, and 0x34 is used for LoRaWAN. You can pretty much use any value you like to get some "isolation" from other nodes using different values. I recall reading some Semtech docs that using a different SyncWord from other nodes wont give you a 100% isolation from that nodes, just about 90% (don't recall the reason, maybe interferences).

For SF and BW, the highest the SF and the lower the BW will usually give you longer distance (but lower bitrates). Also cheap xtal oscillators used on many modules makes using BW lower than 62500 difficult. I'm currently using SF 11 and BW 41700 for longer distance on some projects. Also a good antenna helps here.

br!

morganrallen, 2020-01-28:
I'm closing this as it seems the questions have been answered. I've also started a new feature request to outline functionality to give more clarity in what is going on around CRC errors. See #326, as always, PRs most welcome.

closed 2020-01-28 by morganrallen

dmitrygribenchuk, 2021-02-26:
Hello! I have the same issue with enabled CRC and broken packages (some packages are dropped, some - broken)

IoTThinks, 2021-02-27:
If I'm not wrong, in the current code, even when the CRC is invalid, the data is still kept in FIFO for user processing.

We should have a flag to detect if the CRC is invalid.
Read more at https://github.com/sandeepmistry/arduino-LoRa/issues/326#issue-556363810

IoTThinks, 2021-02-27:
According to the code, by right, the code will ignore to read those packets with invalid CRC
https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L699

```
  if ((irqFlags & IRQ_PAYLOAD_CRC_ERROR_MASK) == 0) {

    if ((irqFlags & IRQ_RX_DONE_MASK) != 0) {
      // received a packet
      _packetIndex = 0;

      // read packet length
      int packetLength = _implicitHeaderMode ? readRegister(REG_PAYLOAD_LENGTH) : readRegister(REG_RX_NB_BYTES);

      // set FIFO address to current RX address
      writeRegister(REG_FIFO_ADDR_PTR, readRegister(REG_FIFO_RX_CURRENT_ADDR));

      if (_onReceive) {
        _onReceive(packetLength);
      }
    }
    else if ((irqFlags & IRQ_TX_DONE_MASK) != 0) {
      if (_onTxDone) {
        _onTxDone();
      }
    }
  }
}
```

ramakantkr, 2021-08-05:
sender and receiver assign different sf ..sf 7,sf 8......data is not received by receiver...even I want receive garbage msg by using different sf in signal modem channel ...thank adv