sandeepmistry/arduino-LoRa #267: RSSI Calculation

manvendra88, 2019-07-11:
RSSI calculation of our library seems to be a little different from what is there in the specs. 
Is there a huge precision difference between the two calculations?
https://github.com/Lora-net/LoRaMac-node/issues/420 : This library uses almost similar RSSI calculations. Should we too? If not, is the difference significant?

torntrousers, 2019-07-12:
If you look in the datasheet (eg [here](https://www.mouser.com/ds/2/761/sx1276-1278113.pdf)) in section 5.5.5 it gives the formula for RSSI which is the same as that used in [this code](https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L257-L260).

I think the code you reference in that Lora-net issue is about calculating the signal strength which is different from the RSSI value and is described further on in section 5.5.5 in the datasheet.

Maybe this library could also add an additional function for getting the signal strength?

manvendra88, 2019-07-13:
Thanks for the clarification. 
In the note, the last line reads like :RSSI = -157+16/15 * PacketRssi (or RSSI = -164+16/15 * PacketRssi) for cases when SNR>0. 
I will add the function for packet strength and PR, for RSSI should we also include the case for SNR>0.

Correct me if wrong about RSSI for SNR>0.

scott2b, 2019-09-09:
I am also reading the spec to mean that the calculation is different for snr > 0 (see especially note 3 of 5.5.5). Furthermore, the packetSnr calculation seems to be hard-coded to the 0.25 slope adjustment that would be used to calculate the packet strength for cases when snr < 0. It is not clear to me why this adjustment is included in the raw SNR calculation. As-is, in order to calculate packet strength for snr > 0, one needs to divide the 0.25 back out of the snr value and adjust by 16/15 instead.

Another furthermore: the hardcoding of the -157 and -164 adjustment of RSSI makes assumptions about the front end configuration (see note 2). Looking at 5.5.5 of the HopeRF module (for example), it's not entirely clear, but I get the impression that the adjustment would be -137 instead: https://cdn-learn.adafruit.com/assets/assets/000/031/659/original/RFM95_96_97_98W.pdf?1460518717