sandeepmistry/arduino-LoRa #268: Please label it properly

Garfius, 2019-07-11:
Once long tested the code, and seen so many issues, and understanding the hardships of the general prupose. I understand this code has no possible professional application.

Therefore, i sincerely ask to modify readme.md to show a warning message regarding a demonstration prupose.

Thanks.

torntrousers, 2019-07-12:
It works well for me. Would you list out the issues you've seen?

Garfius, 2019-07-12:
The most important:

Can't work nore than a few hours.

Interrupt driven receive method is not reliable.

mchacher, 2019-07-12:
Please read following issues. I did the patch on my side.
    // https://github.com/sandeepmistry/arduino-LoRa/issues/218
    // https://github.com/sandeepmistry/arduino-LoRa/issues/222
Since the callback is done under interupt, limit as much as possible the code in the ISR.

torntrousers, 2019-07-12:
If you've got a patch for that issue would you open a PR or show the diff?

mchacher, 2019-07-12:
Just comment line 654 as explained here.
https://github.com/mchacher/LoRa2MQTT/commit/f03fc0bf020fc4ee5898b92cff39429a2ec926a0

mchacher, 2019-07-12:
@torntrousers , just comment line :
_writeRegister(REG_FIFO_ADDR_PTR, 0);_
at end of the function void LoRaClass::handleDio0Rise()

closed 2019-07-12 by morganrallen