sandeepmistry/arduino-LoRa #271: TTGO Lora32 freezes swapping rx/tx modes (ESP32)

mxbranson, 2019-07-18:
Hi,
Trying to receive a packet and send an ack back.
I can send from one device and receive on another ok.
After receiving the packet in onReceive()'s function, I try to lora.beginpacket() then I get a crash and reboot.
If I introduce a delay(2) before hand, I can send the packet (sender gets ack) but the device freezes and doesnt come back from the library. (I have a Serial.print() after sending ack packet that is never displayed).
Device no longer receives any packets nor crashes and reboots.
A reset gets one packet, acks and freezes again.
Any clues?
I tried your examples lora_txMode() and rxMode() around sending a packet with the same results.
Cheers!

morganrallen, 2019-07-18:
Can you posts gists of the code to recreate this? I suspect the next release will fix this and I'm trying to get that out this weekend.

markusschweitzer, 2020-08-19:
I'm having the same issue in the current git version. When i call beginPacket while I'm in the receive callback, it completely freezes :(
Have you figured it out?