sandeepmistry/arduino-LoRa #272: RFM95 M2M RN2903/2843 fail

TomasAllegrini, 2019-07-19:
Hello, i am having the following issue, and i am hoping you can help me
I run the examples and they work perfectly between 2 RMF95, they can both send and receive messages, but what i am trying to achieve is to communicate the RFM95 with a Microchip RN2903/2843. I specify for both LoRa modules, the same parameters, frequency (915Mhz), bandwith (125000), cr (4/5), spreading factor (12) and sync word (using the microchip's default one, 0x39) but i can't make them talk each other.
Thanks! greetings from Argentina!

morganrallen, 2019-07-19:
This driver doesn't support those Microchip devices. This driver communicates directly with the SX127x chip, which the RN2903 contains inside it's 'can'. Additionally it contains a separate microcontroller that you communicate with over UART.

This doesn't mean you cannot use this driver to communicate **with** a RN2903 device, it just won't be able to control one.

closed 2019-07-19 by morganrallen

TomasAllegrini, 2019-07-19:
yes, i know, the RN2903 is working with its radio commands, and the rfm95 with the LoRa library. I could make them talk to each other, but i don't know why, for each byte that i send with the RFM, the RN2903 receive it, but with an additional character, for example, if i send 11 with the rfm, using LoRa.print() i recieve in the RN 3131.
I tried to send the data as a string "11", or using the formats of the LoRa.print, such as HEX

morganrallen, 2019-07-23:
Does this only happen in one direction? RFM>RN2903? It look like you (or the RN2903) might just be processing the incoming data incorrectly. The number `1` is `0x31` in hex.

TomasAllegrini, 2019-07-23:
It has Belén already resolved, they Microchip RN2903 had a diferent
preamble Word and sync than they rfm95, i also had to specify they crc on
both devices yo be on ir off.

TomasAllegrini, 2019-07-23:
Sorry, the previous email was editted by ther autocorrect, it's working
now, i am going to upload the code i made to establish a connection between
both devices

morganrallen, 2019-07-23:
Great, glad you sorted it out.

vleroylr, 2021-09-30:
> […](#)

Did you upload the working code?, I can't seem to find it on your GitHub.
Subiste el código funcionando?, no puedo encontrarlo en tu GitHub.

I'm in the same situation but can't make it work!
Estoy en la misma situación pero no logro hacerlo funcionar!

Thank you, Gracias.