sandeepmistry/arduino-LoRa #274: Transmission takes more than a second

cpsinghopl, 2019-07-25:
Hello,

I checked using oscilloscope, the LoRa module SX1276 takes more than one second to transmit data.

i.e. after following statement is executed (top go to transmit mode), it takes around 100mA for more than second which in my opinion is too much. 

writeRegister(REG_OP_MODE, MODE_LONG_RANGE_MODE | MODE_TX); 

Any idea how to reduce it? with this timings, it is too inefficient to run on battery

PS: Following configuration is used:
  LoRa.begin(868E6);
  LoRa.setSpreadingFactor(12);
  LoRa.setCodingRate4(8);
  LoRa.setSyncWord(0x13);
  LoRa.setPreambleLength(8);
  LoRa.setTxPower(40);
  LoRa.setSignalBandwidth(125E3);
  LoRa.enableCrc();

Regards

torntrousers, 2019-07-25:
The time a transmission takes depends on the spreading factor, bandwidth and coding rate. Its not super speedy and with high spreading factors and low bandwidths it can take several seconds. All you can do is send less data or use a lower spreading factor / higher bandwidth.

cpsinghopl, 2019-07-25:
oh yes, thats what I found now.. on increasing bandwidth from 250 to 500, tx time just halved!
Thanks a lot for your response.

cpsinghopl, 2019-07-25:
But with higher bandwidths, range will be decreased?

IoTThinks, 2019-08-01:
LoRa.setTxPower(40);
Tx power can be so high? Or you are using another more powerful chip than Ra01, 02 and RFM95?

IoTThinks, 2019-08-01:
> But with higher bandwidths, range will be decreased?

Higher bandwidth leads to less sensitivity on antenna.
Hence will decrease range.