sandeepmistry/arduino-LoRa #278: ESP32 with multiple LoRa SX1278 as SPI slave

IoTThinks, 2019-08-01:
Hi,
Thanks a lot for your awesome LoRa library.
I would like to explore how to use esp32 as master and 2 sx1278 as slave.

[image]

The pin mapping as bellow.
```
// LoRa 1 pins
#define LORA_SS          23
#define LORA_SCK         18
#define LORA_MOSI         5
#define LORA_MISO        36
#define LORA_DIO012      39
#define LORA_RST         13

// LoRa 2 pins
#define LORA2_SS         32
#define LORA2_SCK        18 // Shared
#define LORA2_MOSI        5 // Shared
#define LORA2_MISO       36 // Shared
#define LORA2_DIO012     34
#define LORA2_RST        15
```

In Lora 1, I put as bellow
```
  SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_SS);
  LoRa.setPins(LORA_SS, LORA_RST, LORA_DIO012);
  
  while (!LoRa.begin(433E6)) {
    Serial.println("[LoRa] Starting LoRa failed!");
    delay(1000);
  }
```

In LoRa 2, I put as bellow.
```
  LoRaClass LoRa2;
  SPI.begin(LORA2_SCK, LORA2_MISO, LORA2_MOSI, LORA2_SS);
  LoRa2.setPins(LORA2_SS, LORA2_RST, LORA2_DIO012);
  
  while (!LoRa2.begin(433E6)) {
    Serial.println("[LoRa2] Starting LoRa failed!");
    delay(1000);
  }

```
I know I should control the SS (Slave select) but don't know how to integrate it with your library.
Thanks a lot for any advice.

slavino, 2019-08-01:
Maybe offtopic but would this help you any further? https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino

IoTThinks, 2019-08-01:
> Maybe offtopic but would this help you any further? https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino

Thanks.
I read this url already. This one is 2 SPI masters (HSPI and vSPI).

My case is 1 SPI master and 2 LoRa SPI slave.
BTW, I'm still playing around with SS for each slave.

IoTThinks, 2019-08-01:
I set something like this.
I remove SS pin in SPI.begin.
Seems to work but one LoRa by another LoRa.
Not so concurrently.

LoRa 1
```
  SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI);
  LoRa.setPins(LORA_SS, LORA_RST, LORA_DIO012);
  
  while (!LoRa.begin(433E6)) {
    Serial.println("[LoRa] Starting LoRa failed!");
    delay(1000);
  }
```
LoRa 2:
```
  SPI.begin(LORA2_SCK, LORA2_MISO, LORA2_MOSI);
  LoRa2.setPins(LORA2_SS, LORA2_RST, LORA2_DIO012);
  
  while (!LoRa2.begin(470E6)) {
    Serial.println("[LoRa2] Starting LoRa failed!");
    delay(1000);
  }
```

Then I active and detactive the two LoRa chips between receiving.

```
void setActiveLoRa() {
  pinMode(LORA2_SS, OUTPUT);
  pinMode(LORA_SS, OUTPUT);
  digitalWrite(LORA2_SS, HIGH);
  digitalWrite(LORA_SS, LOW);  
}

void setActiveLoRa2() {
  pinMode(LORA_SS, OUTPUT);
  pinMode(LORA2_SS, OUTPUT);  
  digitalWrite(LORA_SS, HIGH);
  digitalWrite(LORA2_SS, LOW);
}
```

Not so concurrently.
```
    // LoRa 1
    setActiveLoRa(); 
    String loraMessage = receiveLoRaMessage();
    if(loraMessage != "") 
      Serial.println("[LoRa] " + loraMessage);
    delay(50);
    
    // LoRa 2
    setActiveLoRa2();
    String lora2Message = receiveLoRa2Message();
    if(lora2Message != "") 
      Serial.println("[LoRa2] " + lora2Message); 
    delay(50);
```

morganrallen, 2019-08-02:
What is the results of that code you say won't run concurrently? Keep in mind, the LoRa receive functions are blocking, so without some kind of threading you wouldn't be able to achieve concurrency anyhow.

IoTThinks, 2019-08-02:
Thanks a lot.
So there is anyway I let chip LoRa 1 to receive (in blocking mode) and let chip LoRa 2 to start hearing for new packet?
So the goal is 2 LoRa chips can take turn receiving / sending faster?

Will threading in Arduino-ESP32 be helpful?

keithr0, 2019-08-02:
Since the ESP32 has two processors, you could separate your transmit and receive functions that way. there are several tutorials on the web on how to run two processes that way.
This one is pretty basic but gives an introduction to the concepts
https://www.youtube.com/watch?v=k_D_Qu0cgu8

IoTThinks, 2019-08-02:
Ok, let me try.

IoTThinks, 2019-11-18:
I'm using tasks to mitigate the issue.
To close.

closed 2019-11-18 by IoTThinks