sandeepmistry/arduino-LoRa #281: Using ESP32 & Lora with VSPI. System Crashes.

parthibanm, 2019-08-20:
Hello I am using Arduino ESP32, RA02 Thinker Lora. Connected to VSPI I/F
As soon as I do a Lora.begin(433E6) I get the Guru Mediation.
Initializing LoRa Chipset.Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
12:46:02.080 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Any pointers to debug the issue.
Thanks

SPIClass *SPI_VSPI;
static void EventHdlr_LoRaInit(void *Params) {
    Serial.printf("\nInitializing LoRa Chipset.");

    SPI_VSPI = new SPIClass(VSPI);
    SPI_VSPI->begin();
    pinMode(5, OUTPUT); //VSPI SS
    
    
    LoRa.setPins(5, 14);
    LoRa.setSPI(*SPI_VSPI);
    Serial.printf("\nGoing to Begin LoRa");
    
    LORA_ACTIVE = LoRa.begin(433E6);
    return; <== Added this code to narrow down.
    for (int i = 0; i < 100 && !LORA_ACTIVE; i++) {
        LORA_ACTIVE = LoRa.begin(433E6);
        Serial.printf(".");
        delay(500);
    }
}

PLEASE see the [FAQ #1](https://github.com/sandeepmistry/arduino-LoRa#faq) about using [setPins](https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins) **BEFORE** submitting an issue.

morganrallen, 2019-08-20:
If you download the [ESP-IDF](https://github.com/espressif/esp-idf/) directly you can use the `idf.py monitor` tool to get more info. This will provide a complete(ish) stack trace that will help us debug this issue.

parthibanm, 2019-08-21:
Hello MorganrAllen,
This seems to be a SPI issue.  Tried 2 Experiments
On a board with just ESP32 (16MB) with no other components on board.
1. Calling Lora.begin() behaves nicely and return false as expected.
2. Changed the VSPI pins to some random pin values and did SPILora.begin() it fails as Load Prohibited. So problem seems to be in SPI part of the code. I have attached the code and console logs. If this rings any bell.
Will try with ESP-IDF and see, never worked on ESP-IDF, so had to install and see. 
Again Thanks

#define LORA_HW_PINOUT_SCK          10
#define LORA_HW_PINOUT_MISO         11
#define LORA_HW_PINOUT_MOSI         23
#define LORA_HW_PINOUT_RESET        14
#define LORA_HW_PINOUT_SELECT       5
#define LORA_HW_PINOUT_INTERRUPT    2
void setup() {
  // put your setup code here, to run once:

  Serial.begin(115200);
  Serial.printf("\nReading Partition");
  
  ReadPartitionTable();

  if (!SPIFFS.begin(FORMAT_SPIFFS_IF_FAILED)) {
    Serial.println("SPIFFS Mount Failed");
    return;
  }
SPILoRa = new SPIClass(VSPI);
SPILoRa->begin(LORA_HW_PINOUT_SCK, LORA_HW_PINOUT_MISO, LORA_HW_PINOUT_MOSI, LORA_HW_PINOUT_SELECT);

This is the console log:

09:44:47.470 -> Reading Partition
09:44:47.470 -> Reading Partition Table InformationFlash chip size: 16777216Partiton table:
09:44:47.470 -> 0 - 10 - 10000 - 640000 - app0 - 0

09:44:47.470 -> 0 - 11 - 650000 - 640000 - app1 - 0

09:44:47.470 -> 1 - 2 - 9000 - 5000 - nvs - 0

09:44:47.503 -> 1 - 0 - e000 - 2000 - otadata - 0

09:44:47.503 -> 1 - 99 - c90000 - 1000 - eeprom - 0

09:44:47.503 -> 1 - 81 - c91000 - 300000 - ffat - 0

09:44:47.503 -> 1 - 82 - f91000 - 6e000 - spiffs - 0

09:44:47.542 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
09:44:47.542 -> Memory dump at 0x400e4f28: 3321291b 33331b28 28233029
09:44:47.542 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.

morganrallen, 2019-08-21:
This doesn't provide any more detail. Neither of the code snippets you have provided show enough to even compile even if I had the same hardware to test on. You need to use the `idf.py monitor` to get a stack trace that will indicate where the problem is actually occurring.

closed 2019-08-21 by morganrallen

parthibanm, 2019-08-21:
Hello,
Here is the full Code. Modifications of LoraDumpRegister Example part of arduino.

 ESP-IDF installation is going to take time for me, since have to go with learning curve of porting new code there if any.

This is the Full Code.
/*
  LoRa register dump

  This examples shows how to inspect and output the LoRa radio's
  registers on the Serial interface
*/
#include <SPI.h>              // include libraries
#include <LoRa.h>

#define LORA_HW_SYNC_WORD           0xBF
#define LORA_HW_PINOUT_SCK          10
#define LORA_HW_PINOUT_MISO         11
#define LORA_HW_PINOUT_MOSI         23
#define LORA_HW_PINOUT_RESET        14
#define LORA_HW_PINOUT_SELECT       5
#define LORA_HW_PINOUT_INTERRUPT    2

SPIClass * SPILoRa = NULL;
void setup() {
  Serial.begin(115200);               // initialize serial
  while (!Serial);

  Serial.printf("\nInitializing VSPI");
  SPILoRa = new SPIClass(VSPI);
  //SPILoRa->begin(LORA_HW_PINOUT_SCK, LORA_HW_PINOUT_MISO, LORA_HW_PINOUT_MOSI, LORA_HW_PINOUT_SELECT);
  SPILoRa->begin();
  Serial.printf("\nSPI Initialized");

    Serial.printf("\nInitializing LoRa Chipset.");

    pinMode(LORA_HW_PINOUT_SELECT, OUTPUT); //VSPI SS        
    LoRa.setPins(LORA_HW_PINOUT_SELECT, LORA_HW_PINOUT_RESET);
    LoRa.setSPI(*SPILoRa);
    Serial.printf("\nGoing to Begin LoRa");
  
    Serial.printf("LoRa Dump Registers");
  
  // override the default CS, reset, and IRQ pins (optional)
  // LoRa.setPins(7, 6, 1); // set CS, reset, IRQ pin
    LoRa.setPins(LORA_HW_PINOUT_SELECT, LORA_HW_PINOUT_RESET);
  if (!LoRa.begin(433E6)) {         // initialize ratio at 915 MHz
    Serial.printf("LoRa init failed. Check your connections.");
    return;
    //while (true);                   // if failed, do nothing
  }

  LoRa.dumpRegisters(Serial);
}

void loop() {
  LoRa.dumpRegisters(Serial);
  delay(10);

}

This the Error Log:
3:00:06.793 -> Initializing VSPI
13:00:06.793 -> SPI Initialized
13:00:06.793 -> Initializing LoRa Chipset.Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
13:00:06.793 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
13:00:06.793 -> Memory dump at 0x400e9cc0: 0c004136 00f01d02 00004136
13:00:06.793 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
13:00:06.793 -> Memory dump at 0x400e9cc0: 0c004136 00f01d02 00004136