sandeepmistry/arduino-LoRa #283: Add library to Platform.io libraries

trycoon, 2019-08-31:
There are quite a few LoRa libraries in Platform.io list of libraries, but this one is still one of the betters ones and should also be found there I think.

morganrallen, 2019-08-31:
Thanks for the tip, I don't know much about platform.io myself but would look into it.
If you or any other regular around here has experience publishing to platform.io I'd be interested in any advice.

trycoon, 2019-09-01:
It quite well documented here: https://docs.platformio.org/en/latest/librarymanager/creating.html
Example from Adafruit: https://github.com/adafruit/Adafruit_ADS1X15/blob/master/library.properties

Once you leave ArduinoIDE for Platform.io you will never turn back! ;-)

universam1, 2019-09-01:
I second that, platformIO is the way to go now.

dhorner, 2019-09-01:
It seems this is just a bad link issue.  It is platformio.org and NOT platform.io which is an entirely different thing.

morganrallen, 2019-09-01:
OK cool, it does look fairly straight forward and I support the addition. @sandeepmistry any input? I'll see about starting this tonight.

sandeepmistry, 2019-09-01:
Sounds ok, it would be ideal if only the library.properties is needed (instead of adding library.json).

From: https://docs.platformio.org/en/latest/userguide/lib/cmd_register.html#id4

Maybe someone just has to register the library.properties URL?

```
platformio lib register https://raw.githubusercontent.com/adafruit/DHT-sensor-library/master/library.properties
```

morganrallen, 2019-09-01:
Great, I will look check that out and likely register this tonight

morganrallen, 2019-09-02:
Welp, turns out, it's already there. Not sure who's publishing it, but there you have it.

https://platformio.org/lib/show/1167/LoRa

closed 2019-09-02 by morganrallen