sandeepmistry/arduino-LoRa #285: SyncWord

AlfieNokes, 2019-09-08:
Hi Am trying to use the LoRa.setSyncWord(syncWord) function

I have set my T-Beam as a sender and a Heltec Board as the receiver and am successfully sending data without issue.

However as I understand it, if I use the setSyncWord function it should only allow my receiver to pick up the data if the sync word set in the transmitter matches that of the receiver.

The problem I have is that I have set a sync word in the transmitter to a different value then the receiver and yet the receiver is picking up the data without a problem.

Could you look into this?

AlfieNokes, 2019-09-08:
Sort of got this working, can the setting parameter be extended above 0xFF

mchacher, 2019-09-08:
Hi @AlfieNokes 
Not sure to understand your second comment _"Sort of got this working, can the setting parameter be extended above 0xFF"_.
I had the same issue and would be interesting if you make it work. Can you be more specific? Thanks.

AlfieNokes, 2019-09-08:
Hi mchacher,

For the transmitter I used the LoRa.setSyncWord function in the setup() block

  LoRa.setSyncWord(0xFF);

for the Receiver I used the same command straight after the Lora.begin

f (!LoRa.begin(BAND)) {
    display.drawString(5,25,"Starting LoRa failed!");
    while (1);
  }
  LoRa.setSyncWord(0xFF);
  Serial.println("LoRa Initial OK!");
  display.drawString(5,25,"LoRa Initializing OK!");
  display.display();
}

my comment regarding extending the parameter is that the syncWord values currently are 0-255.

I believe that its quite possible to extend this to 0-512 or 0-1024 but the library currently only allows 0-255 I would like the developers to maybe extend this function.

If you are still stuck on the sync thing, let me know

hope this helps

Alfie

morganrallen, 2019-09-08:
The SyncWord register is only 8 bits, so at a hardware level only 0-255 are supported. The datasheet shows this at the bottom of the table in section 6.4

Im going to close this, reopen if there is further issue with setting the SyncWord.

https://www.mouser.com/ds/2/761/sx1276-1278113.pdf

closed 2019-09-08 by morganrallen