sandeepmistry/arduino-LoRa #287: Issue during duplex communication

zolag63, 2019-09-14:
Hi, please help me out. 

So i have two arduino nano, connected LoRa sx1278 units. 

I use the to send:
LoRa.beginPacket();
LoRa.write((uint8_t*)&data, sizeof(data));
LoRa.endPacket();

and to receive:
int packetSize = LoRa.parsePacket();
if (packetSize){
    LoRa-readBytes((unit8_t*), packetSize);
}

What works: 
unit1 send to unit 2, unit 2 receive it.
unit 2 send to unit 1, unit 1 receive it.

but if I would like to:
unit1 send to unit 2 than wait for reply - meanwhile unit 2 receive than send

here the is just not works.

the code for the unit1:
#include <SPI.h>
#include <LoRa.h>

byte program_state = 0;
unsigned long pre_millis = 0;

struct payload {
  int id;
  int counter = 0;
  int rnd;
};
typedef struct payload Package;
Package data;

void setup() {
  Serial.begin(115200);
  while (!Serial);
  LoRa.setPins(8, 7);
  Serial.println("LoRa Sender");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
  LoRa.setSpreadingFactor(7);
  LoRa.setSignalBandwidth(62.5E3);
}

void loop() {
  while (1) {
    switch (program_state) {
      case 0:
        if (millis() - pre_millis >= 1500) {
          id();
          tx_msg();
          data.counter++;
          pre_millis = millis();
        }
        program_state = 1;
        break;

      case 1:
        rx_msg();
        program_state = 0;
        break;

      default:
        break;
    }
  }
}

void id() {
  data.id += 1;
  if (data.id >= 5) {
    data.id = 0;
  }
}

void rx_msg() {
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    LoRa.readBytes((uint8_t*)&data, packetSize);
    Serial.print("; rnd: ");
    Serial.println(data.rnd);
  }
}

void tx_msg() {
  LoRa.beginPacket();
  LoRa.write((uint8_t*)&data, sizeof(data));
  LoRa.endPacket();
  Serial.print("Sending packet: ");
  Serial.print(data.counter);
  Serial.print(" to ");
  Serial.println(data.id);
}

unit 2 code:
#include <SPI.h>
#include <LoRa.h>

byte program_state = 0;

struct payload {
  int id = 3;
  int counter = 0;
  int rnd;
};
typedef struct payload Package;
Package data;

void setup() {
  Serial.begin(115200);
  while (!Serial);

  Serial.println("LoRa Receiver");
  LoRa.setPins(8, 7);
  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
  LoRa.setSpreadingFactor(7);
  LoRa.setSignalBandwidth(62.5E3);
}

void loop() {
  while (1) {
    switch (program_state) {
      case 0:
        rx_msg();
        if (data.id == 3)
          prnt_msg();
        program_state = 1;
        break;

      case 1:
        if (data.id == 8) {
          tx_msg();
        }
        data.id = 0;
        program_state = 0;
        break;

      default:
        break;
    }
  }
}

void prnt_msg() {
  Serial.print("Received packet '");
  Serial.print(data.counter);
  Serial.print("' with RSSI ");
  Serial.print(LoRa.packetRssi());
  Serial.print("; SNR: ");
  Serial.println(LoRa.packetSnr());
}

void rx_msg() {
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    LoRa.readBytes((uint8_t*)&data, packetSize);
  }
}

void tx_msg() {
  data.rnd = random(12, 36000);
  Serial.print("; rnd: ");
  Serial.println(data.rnd);
  LoRa.beginPacket();
  LoRa.write((uint8_t*)&data, sizeof(data));
  LoRa.endPacket();
}

What do you think? why I cannot solve this? where is the problem?

zolag63, 2019-09-15:
I found out. at the setPins() I have to declaire the dio0 as wel, so 2 variable not enough, If I don't use the 3rd one, it should be -1
something like this works: LoRa.setPins(8, 7, -1),
this cause issue:                 LoRa.setPins(8, 7);

closed 2019-09-15 by zolag63

universam1, 2019-09-15:
Thanks for updating with the solution