sandeepmistry/arduino-LoRa #289: Duty Cycle Limitations

manvendra88, 2019-09-18:
Does the duty-cycle limitation apply to any network that uses LoRa?
Say, I am using this library in with SX1272 radios to form a network of 10 nodes, should I follow the duty-cycle limitations?

morganrallen, 2019-09-18:
I'm unsure what you're asking, this library does not support SX1272 radios, it will fail the initialization version check.

manvendra88, 2019-09-18:
Hi, I am using the Dragino LoRa shield for Arduino which is based on SX1276/SX1278. 

My question is if I am setting up a network of 5 such nodes and use this library to send sensor data to one transceiver (keeping in mind all the transmitter and receiver have same PHY config), should i obey the duty-cycle limitations of 1%?

manvendra88, 2019-09-18:
I am sorry, it was a typo, I meant SX1276 all along.

morganrallen, 2019-09-19:
Ok that's fine.
I still don't quite understand what you're asking. The duty cycle far as I can tell is set by the radio itself, and is limited to 1%.

There are only a few mentions of duty cycle in the datasheet and the only one the user seems to have control over is for the `PreambleLength` is this what you're referring to? If so, then I *think* I understand and each receiving should have the same or greater PreambleLength set.

From the datasheet section 4.1.1.6.
`The  receiver  undertakes  a  preamble  detection  process  that  periodically  restarts.  For  this  reason  the  preamble  length  should be configured identical to the transmitter preamble length. Where the preamble length is not known, or can vary, the maximum preamble length should be programmed on the receiver side. `

morganrallen, 2019-09-19:
Right, that's also what I was reading. What I don't get from the datasheet is *how* to manage it. Is this some physical thing I'm too RF ignorant to understand?

scott2b, 2019-09-19:
It is my impression that there is not a direct mechanism for regulating the duty cycle physically, aside from simply broadcasting at 18+ dB. You have to track your air time in software and regulate accordingly. How you do this exactly, I'm not yet sure, but would be interested in knowing if you run across anything.

scott2b, 2019-09-19:
Some references that might be useful (have not scrutinized):

https://www.thethingsnetwork.org/forum/t/spreadsheet-for-lora-airtime-calculation/1190

https://www.loratools.nl/#/

I think that second link looks promising

morganrallen, 2019-09-19:
Ah ah ah! I was thinking this was simply the case of being on air, thanks for the clarification.

manvendra88, 2019-09-20:
Thanks for your clarification. 
As for my country, India, I couldn't find any such regulations (Again, I could be wrong too) if you are not using LoRaWAN.For now, I am not taking care of the ToA (Time on Air) duty-cycle limitations of 1% for a small network that I am establishing with 5 tx and 1 Rx talking over same PHY config.

However, if the number of transmitters using the same PHY Config is more than 10 say, then the throughput would surely drop down drastically because we are using a single channel. Here, if we introduce duty-cycle limitation, we can still have the appreciable throughput.

Souls-Bird, 2020-04-29:
Information for European users, ETSI regulations are :
Band 1 -> Freq. 868.0-868.6 MHz | Duty cycle 1% | Max ERP 25 mW / 14 dBm
Band 2 -> Freq. 868.7-869.2 MHz | Duty cycle 0.1% | Max ERP 25 mW / 14 dBm
Band 3 -> Freq. 869.4-869.65 MHz | Duty cycle 10% | Max ERP 500 mW / 27 dBm
Band 4 -> Freq. 869.7-870 MHz | Duty cycle 1% | Max ERP 25 mW / 14 dBm

Source : ETSI TR 103 526, Table 12, Page 35

manvendra88, 2020-05-04:
Hi everyone, I am replying after a long while. Since, now working in a LoRaWAN network in India, SenRa, i think i should answer about the regulations for anybody who will see through this issue. 

India (IN) doesn't have any regulation on the duty-cycle as of yet. However, it is worth noting at this point that, without duty-cycle limitations, a LoRaWAN network with large end-device would for sure have a bad Packet Success Rate (PSR). 

Maybe, in the coming time, Indian bands will also be restricted over the duty-cycle.

Thus, as of now, the limitations on Indian ISM bands is solely on the transmission power emitted and i.e, EIRP<30 dBm.
 
Thank you very much everyone for your time to read and help me solve the issue.

Cheers!

closed 2020-05-04 by manvendra88