sandeepmistry/arduino-LoRa #292: Determining receiver noise floor by accessing radio's RSSI real time value

3gyptian, 2019-09-21:
I've been struggling getting decent range out of my Ai-thinker Ra-01 SX1278 module and found that by adding a function to read register RegRssiValue (0x1b) was extremely helpful to check the health and reception capability at the receiving end without the need of a transmitter. 

This RegRssiValue register provides the immediate rssi value of the receiving radio at a point in time for the bandwidth configured.  During a period of no transmission it is effectively presenting the noise floor of the receiver allowing quick/easy testing of the receiver capability without having to walk around the block to test low signal rssi when receiving packets.  

Here's what I added:

#define REG_RSSI_VALUE           0x1b
...
int LoRaClass::radioRssi()
{
  //return readRegister(REG_RSSI_WIDEBAND); // 4Mhz wideband radio noise floor (random function)
  return readRegister(REG_RSSI_VALUE) - 164 ; // radio noise floor of defined signal bandwidth
}

NOTE: this is for 433Mhz LF module. Subtract 157 for HF.

This showed my receiver to have a noise floor of only -85 db where it should be down closer to -120db.  The main culprits were from me not soldering ALL ground pins on the Ai-Thinker module and not unplugging my laptop from mains power.  Changing those two things along with some others I was able to get my noise floor to around -115db at 62.5 Khz. 

I found graphing continuous samples of the radio rssi also let me quickly see what's going on as I jiggle things or brush the antenna up to things or touch it.  I can also see how the received signal strength varies during an entire transmission period and how it varies over time as I walked around the block. 

[image]

The pulses up to -30 are what I added to show a packet was read.  And the spikes down to -137 are just the first value read from the register after packet received for some reason. 
 
Anyways .. I thought this might be worth sharing with the group and I would be very interested in hearing other people's results with their circuits.

sergio303, 2019-09-22:
Thanks, that is interesting. I'd like to make some tests with this register, how are you recording / plotting values over time ?

3gyptian, 2019-09-25:
To graph the receiver's rssi noise floor I'm sampling the radio's rssi value every 20ms and spitting it out to Arduino Plotter.  

I strongly recommend anyone who isn't getting the distance they expect to perform a noise floor check on their receiver.  Since I posted earlier here is what I now believe allowed me to improve my receiver noise floor from -85db to -115db:

- grounded ALL the pins on the a1 thinker module
- removed hot glue that was touching antenna lead on ai thinker module to adjacent pin and board
- tie wrapping the wires between board and module 

Like this:  

[image]

I now find it doesn't make a difference if my laptop is connected to mains or not .. the noise floor remains steady at -115db.

If anyone's interested here's some receiver code which will dump out to the plotter and also blink the onboard LED to show you how well the receiver is working (assuming nothing being transmitted).   It flashes once representing  every -10 db from 80db and lower (ie: 1 flash = -80-89db, 2 flash = -90-99db, 3 flash = -100-109db etc.  So with nothing being transmitted you want to consistently observe at least 4 flashes representing -110-119db noise floor level.  You can then walk around and see what might be causing interference..or shake your circuit and see how things change.  When a packet is received the output will jump up to -30db for a moment and the onboard LED will flash longer.  And after the first packet is received it will also display the last packet SNR value on the plotter.  Comment out #define FLASH_LED if you don't want your plotter output to pause during analysis.

NOTE: it's using LoRaClass::radioRssi() which I added my local LoRa.cpp and is described in the first post above.

```
#include "LoRa.h" // customized LoRa.cpp to include radioRssi()
#include <FastLED.h> // EVERY_N_MILLIS

#define ledPin 32 // PC13 BluePill STM32F103

#define FLASH_LED

const int csPin = 4;          // LoRa radio chip select
const int resetPin = 17;       // LoRa radio reset
const int irqPin = 16;        // change for your board; must be a hardware interrupt pin

int snr;
int packet_rssi;
long freqErr;
bool receivedpacket=false;

void setup() {
  delay(1000);
  pinMode(ledPin, OUTPUT); // onboard led
  Serial.begin(230400);
  
  // override the default CS, reset, and IRQ pins (optional)
  LoRa.setPins(csPin, resetPin, irqPin);// set CS, reset, IRQ pin

  if (!LoRa.begin(433000000)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
 
  //7.8E3, 10.4E3, 15.6E3, 20.8E3, 31.25E3, 41.7E3, 62.5E3, 125E3, and 250E3.
  //LoRa.setSignalBandwidth(7.8E3);
  //LoRa.setSignalBandwidth(15.6E3);
  //LoRa.setSignalBandwidth(31.25E3);
  LoRa.setSignalBandwidth(62.5E3);
  //LoRa.setSignalBandwidth(125E3);
  LoRa.setSpreadingFactor(12);
  LoRa.setTxPower(20,PA_OUTPUT_PA_BOOST_PIN); // default 17 dbm
  //LoRa.setPreambleLength(60);
  //LoRa.setCodingRate4(5);
  
  Serial.println("Lora radio RSSI readings should be about -115 db or lower for 125KHz bandwidth");

  // Flash LED once at end of setup
  digitalWrite(ledPin, LOW);
  delay(100);
  digitalWrite(ledPin, HIGH);
  delay(800);
}

void loop() {

#ifdef FLASH_LED
  EVERY_N_MILLIS(5000) {
    int radio_rssi=-(LoRa.readRadioRssi()/10);  // one flash for every -10 db from 80db and lower (1 flash=-80db, 2 flash=-90db ...)
    for (int i=8;i <= radio_rssi; i++) {
      digitalWrite(ledPin, LOW); // lights when low
      delay(150);     
      digitalWrite(ledPin, HIGH);
      delay(150);
      int j=LoRa.readRadioRssi(); // throw away noise generated from led pulse
      j=LoRa.readRadioRssi(); // throw away noise generated from led pulse
      j=LoRa.readRadioRssi(); // throw away noise generated from led pulse
      j=LoRa.readRadioRssi(); // throw away noise generated from led pulse
      j=LoRa.readRadioRssi(); // throw away noise generated from led pulse
      }
  }
#endif

  EVERY_N_MILLIS(20) {
    int reg_rssi_value=LoRa.readRadioRssi();
    int reg_lna_gain=LoRa.readLNAGain();
    //Serial.print(reg_lna_gain,HEX);Serial.print(" ");Serial.print(reg_rssi_value); if (receivedpacket) { Serial.print(",");Serial.print(snr);} // don't print snr if haven't received anything - better graphing
    Serial.print(reg_rssi_value); if (receivedpacket) { Serial.print(",");Serial.print(snr);} // don't print snr if haven't received anything - better graphing
    Serial.println();
  }

  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    receivedpacket=true;
    // read the packet
    while (LoRa.available()) {
      int lorachar=LoRa.read();
    }

    packet_rssi = LoRa.packetRssi();
    snr = LoRa.packetSnr();
    freqErr = LoRa.packetFrequencyError();
    
    // spike to steady -30db to show packet received in plotter
     Serial.println("-30");   
     digitalWrite(ledPin, LOW); // lights when low
      delay(500);     
      digitalWrite(ledPin, HIGH);
      delay(100);
  }
}
```

The FASTLed library provides a very convenient EVERY_N_MILLIS do-this timer routine that I use all the time.  And I'm probably going to add some ws2812 LED's to show the radio rssi and packet rssi in real time along with the SNR to help checking distance out in the field.

sergio303, 2019-09-27:
nice! RegRssiValue seems useful to check channel activity and noise floor, I've submitted a PR to add it to the library.

sergio303, 2019-09-27:
I'm already using this to test some antennas and setups I had lying around :)

[image]

3gyptian, 2019-09-27:
That's great .. your noise floor looks in the ballpark.  What is your board/module setup?

I found I was able to drop noise floor an additional 7db down to -122db by tuning the radio frequency for a band that had the least interference.  I modified that receiver code to scan through a range of frequencies while reading the radio rssi and saw there's definitely some noise spikes in my setup every 4.5 Mhz and big spikes every 9 Mhz throughout the receiver's range. I was able to see these on my SDR radio receiver as well with the antenna near the board.   I'm pretty sure this is from the SPI bus and not the CPU as it goes away on my SDR display when the lora module communication pauses as it's flashing the LED.  And my Blue Pill board run's the SPI at 9Mhz..

Here's the serial output of a scan for 430-434 Mhz showing a pretty big interference spike around 432.1 .. 

```
Starting scan...
                 -130                  -120               -110                -100                -90               -80
430.000 -121       ####################
430.015 -118       ##########################
430.030 -120       ######################
430.045 -121       ####################
430.060 -122       ##################
430.075 -121       ####################
430.090 -118       ##########################
430.105 -121       ####################
430.120 -118       ##########################
430.135 -118       ##########################
430.150 -118       ##########################
430.165 -118       ##########################
430.180 -119       ########################
430.195 -118       ##########################
430.210 -118       ##########################
430.225 -118       ##########################
430.240 -118       ##########################
430.255 -118       ##########################
430.270 -118       ##########################
430.285 -117       ############################
430.300 -117       ############################
430.315 -117       ############################
430.330 -117       ############################
430.345 -116       ##############################
430.360 -118       ##########################
430.375 -118       ##########################
430.390 -119       ########################
430.405 -118       ##########################
430.420 -119       ########################
430.435 -119       ########################
430.450 -119       ########################
430.465 -120       ######################
430.480 -118       ##########################
430.495 -117       ############################
430.510 -116       ##############################
430.525 -115       ################################
430.540 -115       ################################
430.555 -112       ######################################
430.570 -111       ########################################
430.585 -111       ########################################
430.600 -111       ########################################
430.615 -112       ######################################
430.630 -111       ########################################
430.645 -110       ##########################################
430.660 -111       ########################################
430.675 -112       ######################################
430.690 -112       ######################################
430.705 -112       ######################################
430.720 -113       ####################################
430.735 -116       ##############################
430.750 -116       ##############################
430.765 -117       ############################
430.780 -119       ########################
430.795 -119       ########################
430.810 -118       ##########################
430.825 -118       ##########################
430.840 -119       ########################
430.855 -119       ########################
430.870 -119       ########################
430.885 -118       ##########################
430.900 -120       ######################
[... 191 lines cut ...]
433.780 -115       ################################
433.795 -114       ##################################
433.810 -113       ####################################
433.825 -113       ####################################
433.840 -113       ####################################
433.855 -113       ####################################
433.870 -113       ####################################
433.885 -113       ####################################
433.900 -114       ##################################
433.915 -114       ##################################
433.930 -114       ##################################
433.945 -115       ################################
433.960 -115       ################################
433.975 -117       ############################
433.990 -117       ############################
Scan done...
```

Which pretty closely matches what the SDR is showing:

[image]

For now at least I've landed at 428.925 Mhz for a steady -122 and sitting fairly wide valley of no interference.

All that said .. there shouldn't be any SPI interference when actually receiving data if instead of continuously polling the module you just wait for a DIO interrupt when packet received.  But then how do you read the noise floor?!

sergio303, 2019-10-01:
I like the scanning idea, will give it a go. My setup is a cheap chinese TTGO lora esp32 board, not great quality, but since mcu,radio and small display are integrated in the same board it's convenient to take it around for measures. 
Since there is activity by serveral nodes on the channel I'm using, it's very useful to use rssi reading to detect channel activity prior to transmission.

3gyptian, 2019-10-08:
Those TTGO boards look pretty great.  I ordered a couple myself.  Definite mini-gateway opportunities with those ESP32 wifi/bt options.

And I'm still deep in my obsession with optimizing RSSI noise floor on my STM32F103 Blue Pills and  AI Thinker RA01 module.  I've learned a lot since I posted above and with some tweaking have now lowered my noise floor consistently down to -130db (and close to SNR ~0) .. believe it or not .. on these wanky little modules.  This is received packet rssi I'm measuring now.   

My next step is to test the optimized receiver in the real world.  If I see matching distance results there I'll share more..

slavino, 2019-10-09:
I found these antennas for 433MHz and those are suitable for real world application. Mage a huge difference just changing these on my TTGO boards.

https://www.tme.eu/sk/details/ant-433-cw-qw-sma/anteny-rf/linx-technologies/

PDF: https://www.tme.eu/Document/d16f332e16434012cdcc9bf9c0aeec50/ant-433-cw-qw.pdf

In my case for long range application:
LoRa.begin(433E6);
LoRa.setSpreadingFactor(10);
LoRa.setSignalBandwidth(62.5E3);

sergio303, 2019-10-28:
Yep, a good antenna is a must. Here cheap chinese antennas are hit and miss, and most ones I've tested are just crap. Unless you have VSWR meter to check them, it's a good idea to stick with good quality brand ones like the Linx Technologies posted above.

IoTThinks, 2020-11-16:
Mentioned in PR https://github.com/sandeepmistry/arduino-LoRa/pull/288/commits

closed 2020-11-16 by IoTThinks

hdrut, 2021-03-01:
> Those TTGO boards look pretty great. I ordered a couple myself. Definite mini-gateway opportunities with those ESP32 wifi/bt options.
>
> And I'm still deep in my obsession with optimizing RSSI noise [...]

Hi, could you share with us some of your techniques to lower the noise floor in your design? 
Thanks!