sandeepmistry/arduino-LoRa #293: Setting spreadfactor and bandwidth doesn't seem to change anything

Nasker, 2019-09-23:
I'm using a TTGO ESP32 LoRa V1 dev board which afaik uses an RF95 LoRa module. Althought it works pretty well when communicating two devices point-to-point, recently I've been trying to tweek the radio parameters to squeeze the data rate. I've tried to set different bandwidth and spreadfactors in order to do so but it doesn't seem to do much neither in the data transmission nor the settings in the API itself (when I call get bandwidth and spreadfactor(by changing those methods to public...)) those remain unchanged.

sergio303, 2019-09-23:
Make sure you're doing the changes after **LoRa.begin**, as this method resets chip and registers to default values.

Nasker, 2019-09-23:
Ooops, that makes perfect sense... I was actually doing those settings before begin. Will try in a couple days and get back to probably close this issue. Thanks!

closed 2020-11-12 by IoTThinks