sandeepmistry/arduino-LoRa #296: ESP-32 with Ai Thinker RA-02. Transmit OK but Receive not working except Library Version 3.0

Cloolalang, 2019-10-08:
Hey Guys.
Im building a ESP-32 & Lora Ra-02 board to use with OpenMqttGateway and Hassio.
I can only get the Lora to receive my test transmitter using Lora.h Version 3.0, not later versions.
The transmit is working perfectly.

I tried the RA-02 on an UNO and got the same results.

Setup:
ESP-32 Dev board
Ai Thinker Ra-02
Lora.h 

RF:
LORA_BAND             434.250E6
LORA_SIGNAL_BANDWIDTH 125E3
LORA_TX_POWER         17
LORA_SPREADING_FACTOR 12
LORA_CODING_RATE      5
LORA_PREAMBLE_LENGTH  5
LORA_SYNC_WORD        0x12 not used from beacon test transmitter
CRC true 

Pins:
#define LORA_SCK     18    
#define LORA_MISO    19   
#define LORA_MOSI    23   
#define LORA_SS      5  
#define LORA_RST     14   
#define LORA_DI0     2  

Any Ideas would be very much appreciated.\
P

slavino, 2019-10-09:
I only used TTGO's boards for LoRa, so may not be much relevant here - but...

FREQ/BAND is 433E6
CRC had to be enabled on both ends (as for transmit so for reception) but the chip was dropping quite a lot from communication then. I decided to switch the CRC off because even corrupt data may serve it's purpose if just small size of packet was damaged.

Spreading factor and signal bandwidth is something you can experiment with for your specific project (I went for spreading 10 and bandwidth 62.5E3 for long range application in woods and fields) with vertical antenna on both ends.

Cloolalang, 2019-10-09:
H

> I only used TTGO's boards for LoRa, so may not be much relevant here - but...
>
> FREQ/BAND is 433E6
> CRC had to be enabled on both ends (as for transmit so for reception) but the chip was dropping [...]
---------------------------------------------------------
Hey Slavino, 
Thanks very much for your comments.
I am using that specific frequency in BAND (434.250 MHz, SRD channel 48) due to the amateur radio and LPD general usage of that band where I assume I will cause the lease interference;

https://www.iaru-r1.org/index.php/spectrum-and-band-plans/uhf/70-centimeter
https://en.wikipedia.org/wiki/LPD433

Thats an interesting point about CRC, I will investigate that as my test 'beacon' has CRC=on.

As for spreading factor, I have tried a few different settings, (not less that 62.5 as I think the Semtech chip requires a stabilised external oscillator for lower spreading factors.

My test beacon has a simple vertical monopole antenna with ground-plane radials at 45 degrees.

I would be keen to know why my receiver works ok on lora.h version 3 or less..
Cheers
P

closed 2021-07-04 by IoTThinks