sandeepmistry/arduino-LoRa #299: Only reply when new message arrives

andresovich, 2019-10-11:
Are you receiving `Starting LoRa failed` while using the demo code?

PLEASE see the [FAQ #1](https://github.com/sandeepmistry/arduino-LoRa#faq) about using [setPins](https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins) **BEFORE** submitting an issue.
Hi everyone, I've been trying to have the code work in the following way:
The sender gathers sensor data and broadcasts it through LoRa, and when a message is received, it sleeps for a while and wakes up using an external RTC. That works almost fine in my setup.

On the receiver side, I am trying to have the code wait for new messages and send a reply when a new one has been received, instead of having it constantly send messages I only mean for it to send a reply after a new message has been received. 

This is the code I've been using. 

`/*RECEIVER
  LoRa Duplex communication
  Sends a message every half second, and polls continually
  for new incoming messages. Implements a one-byte addressing scheme,
  with 0xFF as the broadcast address.
  Uses readString() from Stream class to read payload. The Stream class'
  timeout may affect other functuons, like the radio's callback. For an
  created 28 April 2017
  by Tom Igoe
*/
#include <Arduino.h>
#include <SPI.h>              // include libraries
#include <LoRa.h>
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

// Connecting to WiFi
String readString;
const char* ssid = "Red de H"; // Here you put your netowrks name
const char* password = "antitesis2019"; // Here you put your network's password
const char* host = "script.google.com";//script.google.com:443/HTTPS
const int httpsPort = 443;
WiFiClientSecure client;
const char* fingerprint = "46 B2 C3 44 9C 59 09 8B 01 B6 F8 BD 4C FB 00 74 91 2F EF F6";
String GAS_ID = "AKfycbwMIKZ9DrZBlH9L6glOzDfKe-VBAdVWoK2NL6BQQQUhJoxIC80";  // Replace by your GAS service id. This is the part of the address of your google script that doesn¡t change.

String inmessage;
String incominglength;

const int csPin = 15;          // LoRa radio chip select
const int resetPin = 16;       // LoRa radio reset
const int irqPin = 5;         // change for your board; must be a hardware interrupt pin

String outgoing;              // outgoing message

byte msgCount = 0;            // count of outgoing messages
byte localAddress = 0xFF;     // address of this device
byte destination = 0xBB;      // destination to send to
long lastSendTime = 0;        // last send time
int interval = 2000;          // interval between sends

void sendMessage(String outgoing) {
  LoRa.beginPacket();                   // start packet
  LoRa.write(destination);              // add destination address
  LoRa.write(localAddress);             // add sender address
  LoRa.write(msgCount);                 // add message ID
  LoRa.write(outgoing.length());        // add payload length
  LoRa.print(outgoing);                 // add payload
  LoRa.endPacket();                     // finish packet and send it
  msgCount++;                           // increment message ID
}

void onReceive(int packetSize) {
  if (packetSize == 0) return;          // if there's no packet, return

  // read packet header bytes:
  int recipient = LoRa.read();          // recipient address
  byte sender = LoRa.read();            // sender address
  byte incomingMsgId = LoRa.read();     // incoming msg ID
  byte incomingLength = LoRa.read();    // incoming msg length

  String incoming = "";                 // payload of packet

  while (LoRa.available()) {            // can't use readString() in callback, so
    incoming += (char)LoRa.read();      // add bytes one by one
  }

  if (incomingLength != incoming.length()) {   // check length for error
    Serial.println("error: message length does not match length");
    return;                             // skip rest of function
  }

  // if the recipient isn't this device or broadcast,
  if (recipient != localAddress && recipient != 0xFF) {
    Serial.println("This message is not for me.");
    return;                             // skip rest of function
  }

  // if message is for this device, or broadcast, print details:
  Serial.println("Received from: 0x" + String(sender, HEX));
  Serial.println("Sent to: 0x" + String(recipient, HEX));
  Serial.println("Message ID: " + String(incomingMsgId));
  Serial.println("Message length: " + String(incomingLength));
  Serial.println("Message: " + incoming);
  Serial.println("RSSI: " + String(LoRa.packetRssi()));
  Serial.println("Snr: " + String(LoRa.packetSnr()));
  Serial.println();
  inmessage = packetSize;
}

void setup() {
  Serial.begin(9600);                   // initialize serial
  while (!Serial);

  Serial.println("LoRa Duplex with callback");

  // override the default CS, reset, and IRQ pins (optional)
  LoRa.setPins(csPin, resetPin, irqPin);// set CS, reset, IRQ pin

  if (!LoRa.begin(433E6)) {             // initialize ratio at 915 MHz
    Serial.println("LoRa init failed. Check your connections.");
    while (true);                       // if failed, do nothing
  }

  LoRa.onReceive(onReceive);
  LoRa.receive();
  Serial.println("LoRa init succeeded.");
}

void loop() {

  do {LoRa.receive();
  delay(1000);
  }
  while(LoRa.parsePacket()==0);

  if (millis() - lastSendTime > interval) {
    String message = "HeLoRa World!";   // send a message
    sendMessage(message);
    Serial.println("Sending " + message);
    lastSendTime = millis();            // timestamp the message
    interval = random(2000) + 1000;     // 2-3 seconds
                         // go back into receive mode
  }
}
`

Any help is much appreciated!

levycarneiro, 2022-02-04:
```
void loop() {
  do {
    LoRa.receive();
    delay(1000);
  }
while(LoRa.parsePacket()==0);
(...)

```

Seems the code below is checking for new messages only once every second. You need to be checking / receiving all the time, can't miss a split second otherwise you can miss a message. Don't use delay() as during that 1s wait you might be receiving messages that you are not checking/reading.

The rest is just a bit of logic to achieve what you need. Something like:

- Check for new messages.
- If message is what you expect, then send a reply back.

Good luck.

Kongduino, 2022-02-05:
```c
void loop() {
  do {
    LoRa.receive();
    delay(1000);
  }
while(LoRa.parsePacket()==0);
```

That piece of code doesn't make sense. `LoRa.receive();` shouldn't be called repeatedly. You call it once in `setup()` and shouldn't call it again unless you send something.
The delay is meaningless and haful, as it blocks other threads. And `LoRa.parsePacket()` has nothing to do there either.

Your `loop()` event shouldn't concern itself with LoRa packets reception, that's what `onReceive` is for. Only the random interval but should be in there.