sandeepmistry/arduino-LoRa #306: Duplex extend transmission time

tstanescu1, 2019-11-22:
First of all, thank you all for creating such a straightforward LoRa library, its really great!

I'm having an issue with the duplex, where not all packets get received from one LoRa device to another. I'm using it to control a relay, and some of the commands get dropped. I understand that when sending, the lora can't receive and vice versa. I was wondering if there is a command I could use to extend the air time of the packet in order to let the device loop through the code completely.

 I've attempted to change the SF up to 11 but I didnt seem to see a difference. 
Thanks in advance! Tiberius

IoTThinks, 2019-11-27:
I may not get what you mean.
It seems that you may need Receive callback to quickly go back to receive mode after sending a packet.
https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaDuplexCallback/LoRaDuplexCallback.ino

Or you may need non-blocking sending packet to go gack to do other tasks immediately after sending a packet.
https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaSenderNonBlocking/LoRaSenderNonBlocking.ino#L32

IoTThinks, 2019-11-27:
I believe for packet confirmation / Acknowdgement, you need to write it yourself.
Just reply "ok" with request id.

tstanescu1, 2019-11-27:
Thanks for your reply!

Could you please explain a bit how you would implement the receiver acknowledgment code on the receiver side? Would you add an if statement that would match the msg id? Or without the Ok the message would be sent again? I'd like it to be as seamless as possible.

I'm not very seasoned at programming.. yet. But with your help we could add an additional example in the library to confirm packet receipt. Thanks again!

IoTThinks, 2019-11-28:
Sorry, I have no example for that.
But the logic will be:
Sender to send id:xx,msg:1234
Receiver to send back ack id:xx,msg:ok

If the id is matched, then Sender knows the message xx is received by receiver.

tstanescu1, 2019-11-28:
Thanks! I will see if I can implement it to my code, and add it to the examples list. Cheers!

closed 2019-12-04 by tstanescu1