sandeepmistry/arduino-LoRa #307: No packet received on receiver LoRa-02. Everything worked perfectly fine before

Benedict-Bsl, 2019-11-28:
I have been testing LoRa-02 and using it for quite some time now. But recently i came across a bug which is some how tricky. 
i am using the "LoRaSender" and "LoRaReceiver" example code available in the library to send the "hello word + counter" packet and receiving it, both the receiver and transmitter at a frequency of 433mhz changed in the code from LoRa.begin(915E6) to LoRa.begin(433E6). 

Everything worked perfectly fine before but a couple of days ago it just stopped receiving.... like nothing.

**Sender code:**

`#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(2000);
}`

**Receiver Code**

`#include <SPI.h>
#include <LoRa.h>

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Receiver");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");

    // read packet
    while (LoRa.available()) {
      Serial.print((char)LoRa.read());
    }

    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }
}`

Your help will greatly be appreciated. The library is Fantastic by the way!

IoTThinks, 2019-12-09:
Hi,
You can open an UART terminal to check if the LoRa setup succeeds.
If it prints "Serial.println("Starting LoRa failed!");" then something is failed.

Janadhii, 2020-01-27:
Hello, 
Did you find any solution? I also have the same problem of not receiving packets. The Sender works fine !
Thanks.

NgHieuPickon, 2020-02-17:
Hello, I fixed it.
In funchion endPacket (LoRa.ccp):

```
int LoRaClass::endPacket(bool async)
{
  // put in TX mode
  writeRegister(REG_OP_MODE, MODE_LONG_RANGE_MODE | MODE_TX);

  if (async) {
    // grace time is required for the radio
    delayMicroseconds(150);
  } else {
    // wait for TX done
    while ((readRegister(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0) {
      yield();
    }
    // clear IRQ's
    writeRegister(REG_IRQ_FLAGS, IRQ_TX_DONE_MASK);
  }
  return 1;
}
```

if you trans and receive very fast, the program will stop in "while" => break it!

i have edited that: 

```
  if (async) {
    // grace time is required for the radio
    delayMicroseconds(150);
  } else {
    // wait for TX done
	int fix = 0;
    while ((readRegister(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0) {
      yield();
	  Serial.println("hehe");
	  fix++;
	  delay(1);
	  if(fix > 100){
		  writeRegister(REG_IRQ_FLAGS, IRQ_TX_DONE_MASK);
		  return 0;
	  }
    }
    // clear IRQ's
    writeRegister(REG_IRQ_FLAGS, IRQ_TX_DONE_MASK);
  }
```

And in Arduino:
```
          LoRa_txMode();          
          LoRa.beginPacket();          
          LoRa.print(Res);          // Res is message, type String.
          if (LoRa.endPacket() == 0) {
            Serial.println("error endPacket!");  //print error
          }          
          LoRa_rxMode();
```

English isn't my first language, so please excuse my mistakes!

orimate, 2021-02-15:
Hi, are LoRa_txMode(); LoRa_rxMode() your own functions? Should be not called after LoRa.print(Res); calling this LoRa.endPacket(); ?

closed 2021-07-04 by IoTThinks