sandeepmistry/arduino-LoRa #308: MKRWAN 1310 using SPI flash.

03sarath, 2019-12-02:
Hey sandeepmistry,

 Just now I got my new **MKR WAN 1310** boards from Arduino Store. I'm stuck in using an external  **SPI FLASH 2MB**.I'm using [Marzogh](https://github.com/Marzogh/SPIMemory) library for accessing SPI FLASH chip, I'm a bit confused in searching for **Slave Select pin**, I don't know which pin I have to use as Slave Select for accessing SPI flash.

According to schematic FLASH, chip has been connected to PIN `PB08` on SAMD21

facchinm, 2019-12-02:
Hi @03sarath ,
you can access the external flash using pin 32 as CS https://github.com/arduino/ArduinoCore-samd/blob/master/variants/mkrwan1300/variant.h#L220  and `SPI1` as bus

closed 2019-12-02 by facchinm

disk91, 2022-11-03:
You basically need more than this. I've made a custom firmware to be able to deactivate the UART on lorawan modem without having to reset it.
https://github.com/disk91/mkr1310_openLoRaModem_fw_update