sandeepmistry/arduino-LoRa #309: Lora 's packetLength?

panosss, 2019-12-02:
I 've built a receiver with Arduino which uses the Lora library.
With these settings:
```
frequency = 433E6;
SpreadingFactor = 12;
Bandwidth = 62.5E3;// 500E3;
CodingRate4 =4;
power=20;
SyncWord = 0xF3;
```

I 'm trying to make a sender based on STM32 that will send to the above mentioned receiver.
One of the parameters for begin() is packetLength;
So I  want to set the packet length of the receiver the same as in sender (10).
To be sure that there won't be any problem.

How do I set the packetLength in Lora library?

IoTThinks, 2019-12-09:
Why you want that?
Normally, we need packetLength so that receiver knows exact length of the packet.
Without exact packetLength, the receiver doesn't know where the packet ends.

panosss, 2019-12-14:
(I apologise for the delayed answer, I just now saw it)
I need it because in the code it 's used, so I have to use it too.
```

int SX1278_LoRaEntryRx(SX1278_t * module, uint8_t length, uint32_t timeout) {
	uint8_t addr;

	module->packetLength = length;

	SX1278_defaultConfig(module);		//Setting base parameter
```

IoTThinks, 2019-12-21:
If I understand your question correctly, you want to get the packetLength.

You can use callback. In callback method, you will have packetLength.
https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaReceiverCallback/LoRaReceiverCallback.ino#L30

szotsaki, 2020-01-10:
> If I understand your question correctly, you want to get the packetLength.

Also an option to use my fork at https://github.com/szotsaki/arduino-LoRa. I have that API public under the name of `getPayloadLength()`.

closed 2020-11-12 by IoTThinks