sandeepmistry/arduino-LoRa #310: How to access SPI FLASH & battery charging mode in MKRWAN 1310.

johnnyptpl, 2019-12-08:
Hi.

This is Johnny, I'm exploring new MKRWAN 1310 very low power board. Thanks for Arduino for introduction this board. 

I'm trying to access SPI FLASH &  Battery charging mode, I can see that [https://github.com/arduino/ArduinoCore-samd/blob/2ca7f7531d4e4e54068bc584503b9e02e9033f3b/variants/mkrwan1300/variant.h#L220](url) SPI FLASH is using **PIN 32** as **CS**. But I also really don't know how to activate the Battery charging mode to get battery capacity values.

I tried to access SPI FLASH using` SPI1` bus by the support of [Marzogh](https://github.com/Marzogh/SPIMemory) library but I couldn't access. Here is my code.

```
#include<SPIMemory.h>

uint32_t strAddr;

#if defined(ARDUINO_SAMD_ZERO) && defined(SERIAL_PORT_USBVIRTUAL)
// Required for Serial on Zero based boards
#define Serial SERIAL_PORT_USBVIRTUAL
#endif

#if defined (SIMBLEE)
#define BAUD_RATE 250000
#define RANDPIN 1
#else
#define BAUD_RATE 115200
#define RANDPIN A0
#endif

SPIFlash flash(32, &SPI1);       //Use this constructor if using an SPI bus other than the default SPI. Only works with chips with more than one hardware SPI bus
//SPIFlash flash(32);

bool readSerialStr(String &inputStr);

void setup() {
  SPI1.begin();
  Serial.begin(BAUD_RATE);
#if defined (ARDUINO_SAMD_ZERO) || (__AVR_ATmega32U4__)
  while (!Serial) ; // Wait for Serial monitor to open
#endif

  flash.begin();

  randomSeed(analogRead(RANDPIN));
  strAddr = random(0, flash.getCapacity());
  //strAddr = 4500;
  String inputString = "This is a test String";
  
  //char inArray[6] = {'a', 'b', 'c', 'd', 'e', 'f'};
  //if (flash.writeCharArray(strAddr, inArray, 6)) {
  if (flash.writeStr(strAddr, inputString)) {
    Serial.print(F("Written data: "));
  }
  else {
    Serial.print(F("Unable to write data: "));
  }
  Serial.print(inputString);
  /*for (uint8_t i = 0; i < 6; i++) {
    Serial.print(inArray[i]);
    Serial.print(", ");
  }*/
  Serial.print(F(" to address: 0x"));
  Serial.println(strAddr, HEX);

  String outputString = "";
  char outArray[6];
  //if (flash.readCharArray(strAddr, outArray, 6)) {
  if (flash.readStr(strAddr, outputString)){
    Serial.print(F("Read data: "));
    Serial.print(outputString);
    /*for (uint8_t i = 0; i < 6; i++) {
      Serial.print(outArray[i]);
      Serial.print(", ");
    }*/
    Serial.print(F(" from address: 0x"));
    Serial.println(strAddr, HEX);
  }
  else {
    Serial.println("Unable to read String");
  }
  while (!flash.eraseSector(strAddr));
}

void loop() {

}

//Reads a string from Serial
bool readSerialStr(String &inputStr) {
  if (!Serial)
    Serial.begin(115200);
  while (Serial.available()) {
    inputStr = Serial.readStringUntil('\n');
    Serial.println(inputStr);
    return true;
  }
  return false;
}
```
I have also tried [PaulStoffregen](https://github.com/PaulStoffregen/SerialFlash) library but even that didnt work.Here i have attached the code.

```
#include <SerialFlash.h>
#include <SPI.h>
#define USE_BQ24195L_PMIC
#define FLASH_CS      

const int FlashChipSelect = 32; // digital pin for flash chip CS pin
//const int FlashChipSelect = 21; // Arduino 101 built-in SPI Flash

void setup() {

  SPI1.begin();
  //uncomment these if using Teensy audio shield
 // SPI1.setSCK(14);  // Audio shield has SCK on pin 14
 // SPI1.setMOSI(7);  // Audio shield has MOSI on pin 7

  //uncomment these if you have other SPI chips connected
  //to keep them disabled while using only SerialFlash
  //pinMode(4, INPUT_PULLUP);

  Serial.begin(9600);

  // wait for Arduino Serial Monitor
  while (!Serial) ;
  delay(100);
  Serial.println("All Files on SPI Flash chip:");

  if (!SerialFlash.begin(FlashChipSelect)) {
    error("Unable to access SPI Flash chip");
  }

  SerialFlash.opendir();
  while (1) {
    char filename[64];
    uint32_t filesize;

    if (SerialFlash.readdir(filename, sizeof(filename), filesize)) {
      Serial.print("  ");
      Serial.print(filename);
      spaces(20 - strlen(filename));
      Serial.print("  ");
      Serial.print(filesize);
      Serial.print(" bytes");
      Serial.println();
    } else {
      break; // no more files
    }
  }
}

void spaces(int num) {
  for (int i=0; i < num; i++) {
    Serial.print(" ");
  }
}

void loop() {
}

void error(const char *message) {
  while (1) {
    Serial.println(message);
    delay(2500);
  }
}
```

Serial Monitor Screenshot:
[image]

Requesting help, It will be really supportive if some can provide sample codes for accessing SPI FLASH & activating battery charging mode in MKRWAN 1310

facchinm, 2019-12-09:
Hi @johnnyptpl ,
to access the serial flash chip on MKRWAN1310 you need to reset the LoRa module since its UART is partially shared with SPI1 lanes. The code to do this (using Paul's library) is
```C++
#include <SerialFlash.h>
#include <SPI.h>

const int FlashChipSelect = FLASH_CS; // digital pin for flash chip CS pin

SerialFlashFile file;

const unsigned long testIncrement = 4096;

void setup() {

  Serial.begin(9600);

  while (!Serial) ;
  delay(100);

  pinMode(LORA_RESET, OUTPUT);
  digitalWrite(LORA_RESET, LOW);

  Serial.println("Raw SerialFlash Hardware Test");
  SerialFlash.begin(SPI1, FlashChipSelect); // proceed even if begin() fails

....

```

03sarath, 2019-12-09:
Hey @facchinm,

Thanks in advance.

**My Li-ion battery is not getting Recharged**

1. I'm testing recharging capacity of MKR WAN 1310
2. Trying to read battery status 

 
**1.I'm testing the recharging capacity of MKR WAN 1310**

My MKRWAN 1310 is connected with **Li-Ion battery** along with **USB for external power supply**. But my battery is not getting recharged. Just wondering what is my mistake.

**2. Trying to read battery status** 
I can see in `variant.cpp` of **MKRWIFI 1010**. We have support for reading Battery Status by using `ADC_BATTERY`.

```
void initVariant() {
#if defined(USE_BQ24195L_PMIC)
  pinMode(ADC_BATTERY, OUTPUT);
  digitalWrite(ADC_BATTERY, LOW);
  delay(10);
  pinMode(ADC_BATTERY, INPUT);
  delay(100);

  bool batteryPresent = analogRead(ADC_BATTERY) > 600;
  if (batteryPresent) {
    enable_battery_charging();
  }
  disable_battery_fet(!batteryPresent);
#endif

  // NINA - SPI boot
  pinMode(NINA_GPIO0, OUTPUT);
  digitalWrite(NINA_GPIO0, HIGH);

  // disable NINA
  pinMode(NINA_RESETN, OUTPUT);
  digitalWrite(NINA_RESETN, HIGH);
}

```

**But in MKRWAN 13101 `ADC_BATTERY ` is connected as `FLASH_CS`, Still is that possible to read battery status.**

johnnyptpl, 2019-12-11:
Hello @facchinm,

Thank you, That works. Here I have attached the working code from[ PaulStoffregen ](https://github.com/PaulStoffregen/SerialFlash) 

**From PaulStoffregen** 
```
#include <SerialFlash.h>
#include <SPI.h>

const int FlashChipSelect = FLASH_CS; // digital pin for flash chip CS pin
//const int FlashChipSelect = 21; // Arduino 101 built-in SPI Flash

SerialFlashFile file;

const unsigned long testIncrement = 4096;

void setup() {

  Serial.begin(9600);

  while (!Serial) ;
  delay(100);

  
  pinMode(LORA_RESET, OUTPUT);
  digitalWrite(LORA_RESET, LOW);

  Serial.println("Raw SerialFlash Hardware Test");
  SerialFlash.begin(SPI1, FlashChipSelect); // proceed even if begin() fails

  if (test()) {
    Serial.println();
    Serial.println("All Tests Passed  :-)");
    Serial.println();
    Serial.println("Test data was written to your chip.  You must run");
    Serial.println("EraseEverything before using this chip for files.");
  } else {
    Serial.println();
    Serial.println("Tests Failed  :{");
    Serial.println();
    Serial.println("The flash chip may be left in an improper state.");
    Serial.println("You might need to power cycle to return to normal.");
  }
}

bool test() {
  unsigned char buf[256], sig[256], buf2[8];
  unsigned long address, count, chipsize, blocksize;
  unsigned long usec;
  bool first;

  // Read the chip identification
  Serial.println();
  Serial.println("Read Chip Identification:");
  SerialFlash.readID(buf);
  Serial.print("  JEDEC ID:     ");
  Serial.print(buf[0], HEX);
  Serial.print(" ");
  Serial.print(buf[1], HEX);
  Serial.print(" ");
  Serial.println(buf[2], HEX);
  Serial.print("  Part Nummber: ");
  Serial.println(id2chip(buf));
  Serial.print("  Memory Size:  ");
  chipsize = SerialFlash.capacity(buf);
  Serial.print(chipsize);
  Serial.println(" bytes");
  if (chipsize == 0) return false;
  Serial.print("  Block Size:   ");
  blocksize = SerialFlash.blockSize();
[... 374 lines cut ...]
	return true;
}

bool is_erased(const unsigned char *data, unsigned int len)
{
	while (len > 0) {
		if (*data++ != 255) return false;
		len = len - 1;
	}
	return true;
}

void printbuf(const void *buf, uint32_t len)
{
  const uint8_t *p = (const uint8_t *)buf;
  do {
    unsigned char b = *p++;
    Serial.print(b >> 4, HEX);
    Serial.print(b & 15, HEX);
    //Serial.printf("%02X", *p++);
    Serial.print(" ");
  } while (--len > 0);
  Serial.println();
}
```

closed 2019-12-11 by johnnyptpl