sandeepmistry/arduino-LoRa #315: Implicit header control code

jdliue, 2019-12-21:
From the register description of data sheet, it seems the code should be changed to 
void LoRaClass::explicitHeaderMode()
{
  _implicitHeaderMode = 0;
  writeRegister(REG_MODEM_CONFIG_1, readRegister(REG_MODEM_CONFIG_1) & 0x~~fe~~fb);
}
void LoRaClass::implicitHeaderMode()
{
  _implicitHeaderMode = 1;
  writeRegister(REG_MODEM_CONFIG_1, readRegister(REG_MODEM_CONFIG_1) | 0x~~01~~04);
}

matburnham, 2021-03-25:
How so @jdliue? The [SX1278 datasheet](https://www.semtech.com/products/wireless-rf/lora-transceivers/sx1278#download-resources), shows bit 0 as controlling the header mode:

[image]

IoTThinks, 2021-03-26:
@matburnham I think the current source code is correct?
0xfe is 1111 1110. the last bit of the "and" is 0 => explicit mode.
0x01 is 000 0001. The last bit of the "OR" is 1 => implicit mode.

matburnham, 2021-03-26:
It certainly appears to be. I was having trouble getting implicit mode to work, but the fault was elsewhere. This was one of a number of red herrings.

IoTThinks, 2021-03-27:
Likely somewhere else.

closed 2021-04-01 by jdliue