sandeepmistry/arduino-LoRa #316: TTGO ESP32 Sync Word Different Between Transceivers but They Are Still Talking to Each Other

ZebNZ, 2020-01-06:
Hi all,

I have two ESP32s, one I transmit on and one I receive on. I set different Sync Words on the transmitter and receiver to hopefully stop them communicating with each other, but they are still communicating with each other fine!

TX Code:
`//Libraries for LoRa
#include <SPI.h>
#include <LoRa.h>

//Libraries for OLED Display
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

//define the pins used by the LoRa transceiver module
#define SCK 5
#define MISO 19
#define MOSI 27
#define SS 18
#define RST 14
#define DIO0 26

//433E6 for Asia
//866E6 for Europe
//915E6 for North America
#define BAND 915E6

//OLED pins
#define OLED_SDA 4
#define OLED_SCL 15
#define OLED_RST 16
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels

Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RST);

String LoRaData;

void setup() {

  //reset OLED display via software
  pinMode(OLED_RST, OUTPUT);
  digitalWrite(OLED_RST, LOW);
  delay(20);
  digitalWrite(OLED_RST, HIGH);

  //initialize OLED
  Wire.begin(OLED_SDA, OLED_SCL);
  if (!display.begin(SSD1306_SWITCHCAPVCC, 0x3c, false, false)) { // Address 0x3C for 128x32
    Serial.println(F("SSD1306 allocation failed"));
    for (;;); // Don't proceed, loop forever
  }

  display.clearDisplay();
  display.setTextColor(WHITE);
  display.setTextSize(1);
  display.setCursor(0, 0);
  display.print("LORA RECEIVER ");
  display.display();

  //initialize Serial Monitor
  Serial.begin(115200);

  Serial.println("LoRa Receiver Test");

  //SPI LoRa pins
  SPI.begin(SCK, MISO, MOSI, SS);
  //setup LoRa transceiver module
  LoRa.setPins(SS, RST, DIO0);

  if (!LoRa.begin(BAND)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }

  LoRa.setSpreadingFactor(12);
  LoRa.setSyncWord(0x12);           // ranges from 0-0xFF, default 0x34, see API docs

  Serial.println("LoRa Initializing OK!");
  display.setCursor(0, 10);
  display.println("LoRa Initializing OK!");
  display.display();
}

void loop() {

  //try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    //received a packet
    Serial.print("Received packet ");

    //read packet
    while (LoRa.available()) {
      LoRaData = LoRa.readString();
      Serial.print(LoRaData);
    }

    //print RSSI of packet
    int rssi = LoRa.packetRssi();
    Serial.print(" with RSSI ");
    Serial.println(rssi);

    // Dsiplay information
    display.clearDisplay();
    display.setCursor(0, 0);
    display.print("LORA RECEIVER");
    display.setCursor(0, 20);
    display.print("Received packet:");
    display.setCursor(0, 30);
    display.print(LoRaData);
    display.setCursor(0, 40);
    display.print("RSSI:");
    display.setCursor(30, 40);
    display.print(rssi);
    display.display();
  }
}
RX Code:
`//Libraries for LoRa
#include <SPI.h>
#include <LoRa.h>

//Libraries for OLED Display
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

/*DEFINE THE PINS USED BY LORA TRANSCEIVER MODULE*/
#define SCK 5
#define MISO 19
#define MOSI 27
#define SS 18
#define RST 14
#define DIO0 26

/*SET THE CHANNEL AND FREQUENCY OF LORA TRANSCEIVER*/
#define BAND 915E6 //915MHz, E6 channel

/*DEFINE THE PINS USED BY THE OLED SCREEN*/
#define OLED_SDA 4
#define OLED_SCL 15
#define OLED_RST 16
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels

int counter = 0; //packet counter variable

Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RST); //WHAT DOES THIS DO?

void setup() {

  /*RESET OLED DISPLAY VIA SOFTWARE*/
  pinMode(OLED_RST, OUTPUT);
  digitalWrite(OLED_RST, LOW);
  delay(20);
  digitalWrite(OLED_RST, HIGH);

  /*INIALIZE THE OLED DISPLAY*/
  Wire.begin(OLED_SDA, OLED_SCL);
  if (!display.begin(SSD1306_SWITCHCAPVCC, 0x3c, false, false)) { // Address 0x3C for 128x32 OLED
    Serial.println(F("SSD1306 allocation failed"));
    for (;;); // Don't proceed, loop forever
  }

  /*SETUP THE OLED SCREEN AND DISPLAY SOME TEXT*/
  display.clearDisplay(); //clear the OLED display
  display.setTextColor(WHITE); //Set the text colour to white
  display.setTextSize(1); //set the text size
  display.setCursor(0, 0); //start at top left of OLED screen
  display.print("LORA SENDER "); //display title "LORA SENDER"
  display.display(); //WHAT DOES THIS DO?

  /*INITIALIZE THE SERIAL MONITOR*/
  Serial.begin(115200); //set the baud rate to 115200

  /*DEFINE SOME PINS FOR OLED AND LORA MODULE*/
  SPI.begin(SCK, MISO, MOSI, SS);  //SPI LoRa pins
  LoRa.setPins(SS, RST, DIO0);   //setup LoRa transceiver module

  /*START-UP LORA MODULE?*/
  if (!LoRa.begin(BAND)) { //if LoRa does to begin 915E6?
    Serial.println("Starting LoRa failed!"); //Serial.print
    while (1);
  }

  /*SET THE LORA MODULES SPREADING FACTOR*/
  LoRa.setSpreadingFactor(12); //LoRa spreading factor can range from 7 to 12 (Minimize bandwidth and maximize spreading factor to boost link budget. Maximize coding rate to boost reliability)
  LoRa.setSyncWord(0x13);           // ranges from 0-0xFF, default 0x34, see API docs

  /*TELL USER THAT LORA SETUP WAS SUCCESSFUL VIA OLED SCREEN*/
  display.setCursor(0, 10); //OLED screen second line far left
  display.print("LoRa Initializing OK!"); //display on OLED screen
  display.display(); //WHAT DOES THIS DO?
  delay(2000);
}

void loop() {

  /*SEND A LORA PACKET*/
  LoRa.beginPacket(); //begin the LoRa packet
  LoRa.print("hello: "); //include the word "hello" in the packet
  LoRa.print(counter); //include the variiable "counter"
  LoRa.endPacket(); //end and send the LoRa packet

  /*DISPLAY SOME TEXT ABOUT LORA PACKET ON OLED SCREEN*/
  display.clearDisplay(); //clear OLED screen
  display.setCursor(0, 0); //start at top left of screen
  display.println("LORA SENDER"); //display text "LORA SENDER"
  display.setCursor(0, 20); //second line far left of OLED screen
  display.setTextSize(1); //set the OLED screen text size
  display.print("LoRa packet sent."); //display text "LoRa packet sent." on OLED screen
  display.setCursor(0, 30); //third line far left of OLED screen
  display.print("Counter:"); //display text "Counter" on OLED screen
  display.setCursor(50, 30); //third line fifty character across?
  display.print(counter); //display value of "counter" variable on OLED display
  display.display(); //WHAT DOES THIS DO?

  counter++; //add one to "counter" variable

  delay(10000);
}`

Have I done something wrong or have I just misunderstood what Sync Words are?

Thanks

Zeb

IoTThinks, 2020-01-06:
I also found sync word is not reliable.
Actually, two packets with different sync word may interfere with each other, too.
Hence, I use slightly different frequency for diferent LoRa network. Reliable and no interference.

https://lora-developers.semtech.com/knowledge-base/forum/viewthread/48/

ZebNZ, 2020-01-06:
Hi @IoTThinks,

I need something that is totally unique (like a password or group code) so only the ESP32s with that code can talk to each other. I also want it as secure as possible making it hard for people to figure out the code to get into the group. 

Would changing the frequency slightly do this?

Thanks, 

Zeb

morganrallen, 2020-01-06:
> Would changing the frequency slightly do this?

No, anyone with a basic SDR will be able to see what frequency you are on. If you want security you should encrypt your data, neither changing the frequency nor the `SyncWork` will provide *ANY* measure of security.

As for the `SyncWord` not working, this might be a bug, if the SyncWord doesn't match, the radio should drop the packet.

ZebNZ, 2020-01-06:
Hi all,

Thanks @morganrallen and @IoTThinks I will look into both encrypting the data and changing the frequency sightly!

Thanks again,

Zeb

morganrallen, 2020-01-06:
I'm going to leave this open until someone investigates if SyncWords are actually functioning correctly.

ZebNZ, 2020-01-06:
Just checking that if the transmitter Sync Word is "0x12" and the receiver Sync word is "0x13" the packet will not get received I assume?

Thanks,

Zeb

ZebNZ, 2020-01-07:
Okay I have done some testing!

If the Sync Word on the transmitter and receiver are the same the packets get received fine (this is what should happen).

It seems that if your Sync Word ranges between 0x1 and 0x10 **it works fine** (the packet **will not** get received if the receiver Sync Word is 0x4 and the transmitter Sync Word is 0x5).

If your Sync Word ranges between 0x11 and 0x19 and they are similar (like the receiver is 0x12 and transmitter is 0x13) **they interfere!** However if you skip every second Sync Word (in the range from 0x11 and 0x19) it **does not** interfere (so your transmitter might be 0x13 and your receiver 0x15 and it will not interfere).

But from 0x20 to 0x99 it **works as it should** (if the receiver is on 78 and the transmitter on 79 the packet does not get received).

Is it meant to be like this?

Thanks,

Zeb

morganrallen, 2020-01-07:
Interesting test, nice work. 

The datasheet should describe the expected behavior, the thread linked above suggests the sync word should provide "90% isolation", I find it odd that it isn't 100% but it could be things like receive error allow some to get through.

ZebNZ, 2020-01-07:
Thanks @morganrallen! Am I right that it should be 100% isolation if you range from 0x20 to 0x99?

Thanks,

Zeb

morganrallen, 2020-01-07:
I'm not sure. As I don't fully understand the isolation, or lack of I just don't know. If it does turn out to be a simple bit flip here or there, then yes, I would think the further away they are the less likely they are to erroneously collide. The datasheet doesn't mention anything on SyncWord errors so I can only speculate. I'd recommend setting them farther apart and see if this helps.

ZebNZ, 2020-01-08:
Thanks @morganrallen that's great! I will continue to do some testing! 

Thanks to all who have helped! 

Zeb

closed 2020-02-23 by ZebNZ