sandeepmistry/arduino-LoRa #317: LoRa.packetFrequencyError()

escyes, 2020-01-06:
My code is pretty simple
void onReceive(int packetSize) {
	Serial.println(packetSize);
	for (uint8_t i = 0; i < packetSize; i++)
	{
		Serial.print(LoRa.read(), HEX); Serial.print(" ");
	}
	Serial.print(" <--");
	Serial.print("' with RSSI ");
	Serial.print(LoRa.packetRssi());
	Serial.print("' with SNR ");
	Serial.println(LoRa.packetSnr());
	long frecErr = LoRa.packetFrequencyError();
}
when it get a LoRa packet 
my Esp32 
2
10 1 18 1 2C 9D 2 9 89 C6 C7 3F  <--' with RSSI -66' with SNR 8.75
Guru Meditation Error: Core  1 panic'ed (Coprocessor exception)
Core 1 register dump:
PC      : 0x400d0fcb  PS      : 0x00060831  A0      : 0x800d0cd8  A1      : 0x3ffbe650  
A2      : 0x3ffbfd40  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000002  A7      : 0x00000085  A8      : 0x800d0fcb  A9      : 0x3ffbe6f0  
A10     : 0x0001e848  A11     : 0x00000028  A12     : 0x00000000  A13     : 0x40218000  
A14     : 0x00000002  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000004  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
Core 1 was running in ISR context:
EPC1    : 0x400d0fcb  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400826e9

Backtrace: 0x400d0fcb:0x3ffbe650 0x400d0cd5:0x3ffbe710 0x400d0cd5:0x3ffbe740 0x400d1433:0x3ffbe760 0x40080e3e:0x3ffbe780 0x40080e89:0x3ffbe7a0 0x40081671:0x3ffbe7c0 0x400ea035:0x3ffb1fb0 0x40087921:0x3ffb1fd0

It comes from LoRa.packetFrequencyError(), if  isn't there all go fine.

¿any idea?

morganrallen, 2020-01-06:
What device is this on?

escyes, 2020-01-07:
Thank you very much for your interest, I assumed that this problem could be general, I use an Esp32 as motherboard
LOLIN D32 Pro V2.0.0
And my LoRa module is Modtronic inAir9 that uses sx1276 connected by RFO
I am going to try to make changes in the pins that I can change, in case there is any conflict, but it is surprising that everything working well only breaks with that function.
I am greatly grateful for your work, I am currently migrating from other more complex libraries because I have more control.

torntrousers, 2020-01-07:
Could you try installing the Exception Decoder so we can see exactly what line its at? 
https://github.com/me-no-dev/EspExceptionDecoder

escyes, 2020-01-07:
PC: 0x400d0fcb: LoRaClass::packetFrequencyError() at C:\Users\escye\Documents\Arduino\libraries\LoRa\src\LoRa.cpp line 283
EXCVADDR: 0x00000000

Decoding stack results
0x400d0fcb: LoRaClass::packetFrequencyError() at C:\Users\escye\Documents\Arduino\libraries\LoRa\src\LoRa.cpp line 283
0x400d0cd5: onReceive(int) at C:\Users\escye\Documents\GpsCowBell_2020\LoRaReceiver_sandeepmistry/LoRaReceiver_sandeepmistry.ino line 14
0x400d0cd5: onReceive(int) at C:\Users\escye\Documents\GpsCowBell_2020\LoRaReceiver_sandeepmistry/LoRaReceiver_sandeepmistry.ino line 14
0x400d1433: LoRaClass::handleDio0Rise() at C:\Users\escye\Documents\Arduino\libraries\LoRa\src\LoRa.cpp line 653
0x40080e3e: LoRaClass::onDio0Rise() at C:\Users\escye\Documents\Arduino\libraries\LoRa\src\LoRa.cpp line 686
0x40080e89: __onPinInterrupt at C:\Users\escye\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\esp32-hal-gpio.c line 220
0x400d224b: loopTask(void*) at C:\Users\escye\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\main.cpp line 14
0x40087921: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

torntrousers, 2020-01-07:
Which is pointing at this line:
https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L283

escyes, 2020-01-07:
I'working in it.

torntrousers, 2020-01-07:
Tremendous. I don't know what the problem is or how you're thinking of fixing it, but its not ideal that the coding is using floating point arithmetic on those last three lines. Fixing that might make the crash go away. Its probably possible to refactor it into some integer constant for each bandwidth value and use just integer arithmetic.

escyes, 2020-01-07:
just, but "radiohead" library uses something similar and it worked for me, and it was congruent.
You're right, I'm trying to rewrite the code and it's in that cast to float were is the problem
This fails:
int LoRaClass::frequencyError()
{
    int32_t freqerror = 0;

    // Convert 2.5 bytes (5 nibbles, 20 bits) to 32 bit signed int
    freqerror = static_cast<int32_t>(readRegister(REG_FREQ_ERROR_MSB)) << 16;
    freqerror |= static_cast<int32_t>(readRegister(REG_FREQ_ERROR_MID)) << 8;
    freqerror |= static_cast<int32_t>(readRegister(REG_FREQ_ERROR_LSB));
    // Sign extension into top 3 nibbles
    if (freqerror & 0x80000)
        freqerror |= 0xfff00000;

    int error = 0; // In hertz
    //float bw_tab[] = { 7.8, 10.4, 15.6, 20.8, 31.25, 41.7, 62.5, 125, 250, 500 };
    //uint8_t bwindex = spiRead(RH_RF95_REG_1D_MODEM_CONFIG1) >> 4;
    //if (bwindex < (sizeof(bw_tab) / sizeof(float)))
    error = (int)((float)freqerror);//* bw_tab[bwindex] * ((float)(1L << 24) / (float)RH_RF95_FXOSC / 500.0);
    // else not defined

    return error;
}
Just this (float)freqerror) 
but the funny is that:
(float)123456L works 
NOW i'm thinking how to avoid float

IoTThinks, 2020-03-08:
This method runs fine for my esp32 with release 0.7.0.

My value is around +/-2000.

IoTThinks, 2020-11-25:
I hit the same issue.
Instead of calling packetFrequencyError  in ISR, need to set Flag and use packetFrequencyError in main loop or spin off a RTOS task to call packetFrequencyError

closed 2020-11-25 by IoTThinks