sandeepmistry/arduino-LoRa #321: Problem with cycles and timing

augusto3691, 2020-01-14:
Hello everyone,

I have a cycle in my nodes thats look like these:
Start listening -> listening for x seconds -> do some stuff -> broadcast the data -> repeat

The problem is that sometimes one node will broadcast info when others not listening. I don't know if my way to think is wrong, if i need to use a onReceive callback and receive it at same time that im sending ,if the window between listening and broadcast is too short, or this lib is not suitable for me. 

Anyone have this issue? Or know the way to help with that ?

IoTThinks, 2020-01-15:
1. You need to use onReceive callback for more continuous listening
2. Once you receive something, you create a new background task to "process other stuff" using FreeRTOS xTaskCreate. If you have multiple core MCU like ESP32, then it's better.
3. Send something and quickly back to listen mode.

The most important thing is step 2. A background task will help us to quickly back to listen mode.
A background task with priority 0 will process data once the MCU is idle.

IoTThinks, 2020-01-15:
I have a sample here.
In my case, I have dual core ESP32 with two LoRa SX1278 chips. One to listen and one to send.
If you have 1 SX1278 only, then it should be similar.

https://github.com/IoTThinks/EasyLoRaGateway_v2.1/blob/master/EasyLoRaGateway/09_lora.ino#L85
Hope it helps.

This part is to create lower priority task to read and process LoRa message.
```
void onLoRa1ReceiveCallback(int packetSize) {
  xTaskCreate(loRa1ReadTask, "loRa1ReadTask", 10240, (void*)&packetSize, CRONJOB_READLORA_PRIORITY, NULL);
}
```

Once finish sending data, quickly go back to receive mode with LoRa.receive();

augusto3691, 2020-01-15:
OMG i ll really try
i never use the FreeRTOS xTaskCreate and have no idea what is this hahahah Is this a lib ? This is the ESP32 board that im using:

[image]

I tried to use the onReceive with SD card but o always get the Guru Meditation Error but ll work on it.

Thank you very much

IoTThinks, 2020-01-15:
Your board is similar to mine. It's ESP32 + SX1278.
Arduino-esp32 has built-in FreeTROS commands. So you don't need to add/import extra library to use the command.
You can view example at https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/FreeRTOS/FreeRTOS.ino

For your board, you need to check where your DIO0 mapping to.
And set the pins accordingly before using onReceive.
https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins

Hope it helps.

closed 2020-11-12 by IoTThinks