sandeepmistry/arduino-LoRa #322: Receiver Callback Problem

RogerRU, 2020-01-15:
Atmega128A-AU
I use ST7735 SPI display in my project. (ucgib)
The pure Receiver Callback demo example works fine, interrupt works fine, but as soon as I initialize ucglib the interrupt does not triggered.

`#include <SPI.h>
#include <LoRa.h>
#include <Ucglib.h>

#ifdef ARDUINO_SAMD_MKRWAN1300
#error "This example is not compatible with the Arduino MKR WAN 1300 board!"
#endif
#define TFT_CS 33
#define TFT_RST 32
#define TFT_DC 31
#define TFT_BACKLIGHT 14

#define LORA_CS 15
#define LORA_RST 44
#define LORA_IRQ_PIN 6

Ucglib_ST7735_18x128x160_HWSPI tft(TFT_DC, TFT_CS, TFT_RST);

void setup() {
  Serial.begin(9600);
  while (!Serial);
  Serial.println("LoRa Receiver Callback");
  LoRa.setPins(LORA_CS, LORA_RST, LORA_IRQ_PIN);
  if (!LoRa.begin(433E6)){
    Serial.println("Starting LoRa failed!");
    while (1);
  }
  // register the receive callback
  LoRa.onReceive(onReceive);
  // put the radio into receive mode
  LoRa.receive();

  tft.begin(UCG_FONT_MODE_SOLID);
  tft.clearScreen();
  tft.setRotate270();
  analogWrite(TFT_BACKLIGHT, 255);
 
}

void loop() {
  // do nothing
}

void onReceive(int packetSize) {
  // received a packet
  Serial.print("Received packet '");

  // read packet
  for (int i = 0; i < packetSize; i++) {
    Serial.print((char)LoRa.read());
  }

  // print RSSI of packet
  Serial.print("' with RSSI ");
  Serial.print(LoRa.packetRssi());
  Serial.print("' with SNR ");
  Serial.println(LoRa.packetSnr());
}`
Any idea why?

IoTThinks, 2020-01-17:
You may need to specify and start a specific SPI for LoRa.
Otherwise, I guess both of them may use the same SPI interface.

https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-spi-interface

RogerRU, 2020-01-18:
I took the library from Adafruit (Adafruit_ST7735), everything worked fine, BUT the system began to randomly reboot or hang after 2 - 7 hours of work ((((((

IoTThinks, 2020-01-23:
Then you may need to check heap fragementation if you use String.

I use esp32+this lib. Can run in weeks after removing String.

closed 2020-11-12 by IoTThinks