sandeepmistry/arduino-LoRa #323: Stuck program at LoRa.endPacket()

bhaktinandan, 2020-01-21:
i use arduino uno and i use external power supply for RFM96.
at first when i buy, simply test it using transmitter and receiver example, it work like charm.
now after one month again same hardware same code not work. it only send one packet and stuck at LoRa.endPacket(). 
One change i made from statting is i change LORA_DEFAULT_DIO0_PIN to 2 in Lora.h
i also set TX power to 2.

i paste my programme link. please help me.

morganrallen, 2020-01-21:
If you change those back to original does it work again?

bhaktinandan, 2020-01-22:
yes it works.

https://pastebin.com/raw/w2CFyZJs

IoTThinks, 2020-01-22:
LORA_DEFAULT_DIO0_PIN is already 2 in LoRa.h.
So did you really change it?

```
#if defined(ARDUINO_SAMD_MKRWAN1300)
#define LORA_DEFAULT_SPI           SPI1
#define LORA_DEFAULT_SPI_FREQUENCY 200000
#define LORA_DEFAULT_SS_PIN        LORA_IRQ_DUMB
#define LORA_DEFAULT_RESET_PIN     -1
#define LORA_DEFAULT_DIO0_PIN      -1
#elif defined(ARDUINO_SAMD_MKRWAN1310)
#define LORA_DEFAULT_SPI           SPI1
#define LORA_DEFAULT_SPI_FREQUENCY 200000
#define LORA_DEFAULT_SS_PIN        LORA_IRQ_DUMB
#define LORA_DEFAULT_RESET_PIN     -1
#define LORA_DEFAULT_DIO0_PIN      LORA_IRQ
#else
#define LORA_DEFAULT_SPI           SPI
#define LORA_DEFAULT_SPI_FREQUENCY 8E6 
#define LORA_DEFAULT_SS_PIN        10
#define LORA_DEFAULT_RESET_PIN     9
#define LORA_DEFAULT_DIO0_PIN      2
#endif
```

bhaktinandan, 2020-01-23:
Sorry . First time when i test i change it to 8 and again set to 2.

IoTThinks, 2020-01-23:
Then check if the DIO0 of LoRa Sx1278 is connected to GPIO8 of Uno.
And check if GPIO8 of Uno is intterruptable.

bhaktinandan, 2020-01-23:
OK, Right now DIO0 of LoRa Sx1278 is connected to GPIO2 of Uno.
And GPIO2 is intterruptable.

Check my Code. https://pastebin.com/raw/w2CFyZJs

GPIO8  of Uno use previously, now it change to GPIO2 of uno.

IoTThinks, 2020-01-29:
GPIO2 is a special pin in ESP32.
For Uno, you can check the different between GPIO2 and 8.

Silly question. Did you physically plug dio0 of LoRa to GPIO2 of Uno?

bhaktinandan, 2020-01-29:
" Did you physically plug dio0 of LoRa to GPIO2 of Uno? "      = Yes
But wow after reinstall Lora library now it's work.
Thank you so much IoTThinks for Help me..

closed 2020-01-29 by bhaktinandan