sandeepmistry/arduino-LoRa #328: Tx Done Callback Problems

bonedaddy, 2020-01-29:
I'm attempting to use the `onTxDone` callback function, but get the following error:

```

/tmp/arduino_modified_sketch_308599/StandardFirmata.ino: In function 'void setup()':
StandardFirmata:28:10: error: 'class LoRaClass' has no member named 'onTxDone'
     LoRa.onTxDone(onTxDone);
          ^~~~~~~~
Multiple libraries were found for "LoRa.h"
 Used: /home/solidity/Arduino/libraries/LoRa
Multiple libraries were found for "SPI.h"
 Used: /home/solidity/.arduino15/packages/arduino/hardware/avr/1.8.2/libraries/SPI
exit status 1
'class LoRaClass' has no member named 'onTxDone'
```

Here's my sketch:

```
#include <LoRa.h>
/* LibP2P LoRa Transport Arduino Bridge
 * Enables a LibP2P node to send messages over LoRa and possibly LoRaWAN.
 * Data coming in through the serial interface leaves through the LoRa radio
 * Data coming in through the LoRa radio exits the serial interface
 * This is the "fast" version and operates on bytes instead of strings
 */

#define BAND 915E6
#define SYNC 0x99
#define SF 7
#define TXPOWER 20
#define SIGBW 250E3
#define DATASENT 0xd34db33f
/*
API documentation: https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md
*/

void setup() {
    Serial.begin(115200);
    while (!Serial);
    LoRa.begin(BAND);
    LoRa.setSyncWord(SYNC);
    LoRa.setSpreadingFactor(SF);
    LoRa.setTxPower(TXPOWER);
    // register callbacks
    LoRa.onReceive(onReceive);
    LoRa.onTxDone(onTxDone);
    //LoRa.enableCrc();
  //  LoRa.setSignalBandwidth(SIGBW);
    Serial.println("ready");
    Serial.flush();
}

// callback function whenever we receive a LoRa packet
void onReceive(int packetSize) {
  if (packetSize) {
      char buffer[255];
      int num = LoRa.readBytes(buffer, 255);
      Serial.write(buffer, num);
      Serial.flush();
  }
}

// callback function whenever we finished sending a LoRa packet
void onTxDone() {
  Serial.println(DATASENT);
  Serial.flush();
}

void loop() {
  // see if we have any data on the serial interface
  // if we do send it down hte LoRa radio
  if (Serial.available()) {
    if (LoRa.beginPacket()) { // beginPacket returns 1 if ready to proceed, so wait until radio is ready
        char buffer[255]; // 255 byte buffer
        int num = Serial.readBytesUntil('\n', buffer, 255);
        LoRa.write(buffer, num);
        LoRa.endPacket(true); // async mode
        Serial.println(DATASENT);
        Serial.flush();
    } 
  }
}
```

morganrallen, 2020-01-29:
This has yet to be pushed out in a release, so unless you're using the current master branch it would not be available.

closed 2020-01-29 by morganrallen

bonedaddy, 2020-01-29:
How unstable is master at the moment? I'm not doing anything mission critical atm and just experimenting with the platform so possible bugs and stuff aren't the biggest issue.

IoTThinks, 2020-03-08:
@morganrallen: When will the txDone and cadDone be available in the new release. 
Let say v0.8.

Are we pending for more testing?
Thanks a lot.