sandeepmistry/arduino-LoRa #330: how to check lora is working or possible to reset?

halukmy, 2020-02-01:
sometimes when energy comes low or high, lora may stop working, so how can we check lora is working or reset it?

Benedict-Bsl, 2020-02-01:
You can do that by using a pull up resistor attached to a led. (you will have to make sure to select the right one according to the voltage of your led). By using the Lora.available() function(in an if statement) , you can switch on the led when evenever your module isn't working and off in contrary.

Benedict-Bsl, 2020-02-01:
And it's already a mistake for your source of power to come low or high. Use a voltage regulator to keep it constant(3.3v with a minimum amperage of 15mA). And do not forget to use logic level shifter incase you are using a 5v logic board. 
Hope this helps you.

closed 2020-11-12 by IoTThinks