sandeepmistry/arduino-LoRa #332: Some packages are wired

kessero, 2020-02-02:
I'm sending every 5s int = 0 or 1 and for some reason I recive wired data in random packages:
Received packet '48' with RSSI -57<\r>
**<\n>Received packet '18511122419315083862959191872441121731543423847480130211156885637146236121166816520710461759191209161' with RSSI -99<\r>**
<\n>Received packet '49' with RSSI -21
Why I getting this long number in packet data?

Benedict-Bsl, 2020-02-02:
Clarify more on what you are doing exactly. And post the got if wanna be helped.

kessero, 2020-02-02:
LoRa reciver
`#include <SPI.h>
#include <LoRa.h>
#define ALARM_PIN 3
void setup() {
  Serial.begin(9600);
  pinMode(ALARM_PIN, OUTPUT);
  digitalWrite(ALARM_PIN, LOW);      
  while (!Serial);

  Serial.println("LoRa Receiver");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}
char alarm = 0;
void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    alarm = 0;
    // received a packet
    Serial.print("Received packet '");

    // read packet
    while (LoRa.available()) {
      alarm = LoRa.read(); 
      Serial.print(alarm);
           
      
    }

    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }  

  if(alarm == '1' ){
    digitalWrite(ALARM_PIN, HIGH);
  }
  if (alarm == '0'){
    digitalWrite(ALARM_PIN, LOW);    
  }
}`

Lora Sender
`#include <SPI.h>
#include <LoRa.h>

int counter = 0;
int on = 0;
void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}
void send_data(int s){
  Serial.println(s);
  LoRa.beginPacket();
  LoRa.print(s);  
  LoRa.endPacket();
  LoRa.sleep();
  counter++;

  
}
void loop() {
  send_data(0);
  delay(5000);
  send_data(1);
  delay(10000);
  
  
}`

Benedict-Bsl, 2020-02-02:
Try to remove LoRa.sleep();
And post a picture of your connection. 
Then show me the output.

kessero, 2020-02-02:
_Try to remove LoRa.sleep();_
I need low power consumption but i try without this option.
 _And post a picture of your connection._
"Picture of connection"?? What you meen?

Benedict-Bsl, 2020-02-02:
Your connection... Hardware connection

Benedict-Bsl, 2020-02-02:
The schematic diagram of how you have interfaced Lora module with your microcontroller

Benedict-Bsl, 2020-02-02:
LoRa by default was designed as a low power functioning device. I think the reason why you are getting those weird values is that you have to wake up the module after sleeping it. And you do that by reconfiguring LoRa.begin().
Simple way of doing that is checking if the module is available after you made it sleep.

Benedict-Bsl, 2020-02-02:
`void send_data(int s){
if(Lora.available()) {
    Serial.println(s);
    LoRa.beginPacket();
    LoRa.print(s);  
    LoRa.endPacket();
    LoRa.sleep();
    counter++;
else{
  Lora.begin(433E6);
 Serial.println(s);
    LoRa.beginPacket();
    LoRa.print(s);  
    LoRa.endPacket();
    LoRa.sleep();
    counter++;
} 

  
}`

kessero, 2020-02-02:
OK I will test and include your suggestions. Thanks in advance.
I use module like this [https://www.disk91.com/2019/technology/lora/first-steps-with-lora-radio-node-arduino/](url)
So no other hardware for now is in use.

Benedict-Bsl, 2020-02-02:
Let me know if it works. :)

max763, 2020-02-05:
Hi Benedict, I'm sending every second 32bytes to the receiver, which is working fine, but after some time (30min up to 1h) the the receiver doesn't get any message. A reset with LoRa.begin(868e6) helps but how can this status be detected ? LoRa.available() is 0 as the sender doesn't get any message. many Thanks max

kessero, 2020-02-05:
> Let me know if it works. :)

Thanks looks like yours advise help :+1:

Benedict-Bsl, 2020-02-05:
> Hi Benedict, I'm sending every second 32bytes to the receiver, which is working fine, but after some time (30min up to 1h) the the receiver doesn't get any message. A reset with LoRa.begin(868e6) [...]

Hello Max. Please paste the programming code you are actually using. From both of your sender and receiver. And include a picture of your setup(hardware connection of ur lora device and the microcontroller you are using)

Benedict-Bsl, 2020-02-05:
> > Let me know if it works. :)
> 
> Thanks looks like yours advise help

Thank you too. You can now close the issue.

morganrallen, 2020-02-05:
> Hello Max. Please paste the programming code you are actually using. From both of your sender and receiver. And include a picture of your setup(hardware connection of ur lora device and the [...]

Please start a new issue

Benedict-Bsl, 2020-02-05:
Max should do that.

closed 2020-02-05 by kessero