sandeepmistry/arduino-LoRa #335: sending stopped working

max763, 2020-02-05:
I've put up a LoRa class (FLoRa) which has the readloop running in a thread. A callback function can be connected that is called with
the new data received. I added some more functions to LoRa as I found out that I need to increase the receive timeout to get my code working.                            
                                                                                
In the examples of FLoRa (see attached) there is a receiver (FLoRa_receiver) and a sender      
(FLoRa_sender) sketch.                                                          
                                                                                
Every 2sec the sender sends a message and the receiver receives it.             
It is working well in a setup with two ESP32 (RFM95W)  for a quite decent time but after 30minutes or sometimes much more the receiver                    
doesn't get any messages while the sender still claims to send.                 
I've added a button to the sender to reset LoRa with LoRa.begin() etc.          
Once the receiver stops receiving messages it is sufficient to                  
press the button at the sender to initiate the reset. The receiver              
starts receiving again.
I've as well added a button at the receiver to send a message to the looping sender, which is
working fine for a lot of send actions but with pressing a few times the aforementioned
problem occurs earlier than before.
I wonder whether there is a status flag that can be read from the chip to see whether a reset is necessary.                                                            
[FLoRa_max.zip](https://github.com/sandeepmistry/arduino-LoRa/files/4161129/FLoRa_max.zip]                                                        
Thanks and regards                                                              
Max

morganrallen, 2020-02-05:
This provided code isn't that useful, it only contains a copy of the LoRa library (including swap files, suggested it's been edited, therefore not a proper release and unsupportable), FLoRa (what ever that is) and no setup/loop code that actually does anything.

The only thing I can speculate is 2 seconds between sends is too fast, depending on settings. I would recommend you provide a minimal example that can reproduce the issue, otherwise I don't think anyone is going to be able to help wade through multiple wrapper libraries to debug the code.

Additionally several issues that are similar to this have been reported, I would recommend you check through those to see if you can find anything helpful.

max763, 2020-02-05:
sorry for not being clear enough. The two codes are in FLoRa/examples : FLoRa_receiver 
and FLoRa_sender. I had to add void setrxstimeout(uint16_t);  to the LoRa library to increase the RX_timout to make it work for at least an 1h until the sender doesn't send anymore.   The spreadingFactor is set to 7. I thought that 2 seconds should not be a problem, but will increase to check. So far I've rather found issues where receiving stopped working but nothing related to sending. As I'm only have to restart the sender I guess sending is the culprit.

IoTThinks, 2020-02-09:
Too lazy to see your code.
You may copy part of your code here.

If you use String during receiving in receiver, then likely it is heap fragmentation.

Print this regularly in receiving node to check
esp_get_free_heap_size()
esp_get_minimum_free_heap_size()

IoTThinks, 2020-02-09:
Btw this code is prone to heap fragmentation. And the board will be hang after a while.

https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaDuplex/LoRaDuplex.ino#L82

Need to use char[ ] instead of adding String

morganrallen, 2020-02-09:
Also the fact that all the examples do way too much work inside the interrupt doesn't help. See PR #55

closed 2020-12-19 by IoTThinks