sandeepmistry/arduino-LoRa #336: ttgo lora esp32 has so bad rssi values when they are so close why?

halukmy, 2020-02-10:
i have sx1276 ttgo lora, when they close it should be like - 8 or even -15 rssi right?

when i get close  i got like rssi -40 or -50 rssi value, is that normal?

i use everything for make it boost like 

 LoRa.setTxPower(20, PA_OUTPUT_PA_BOOST_PIN );
  LoRa.setSpreadingFactor(10);

anybody can handle it?

i got stock antenna and device working at around 840- 868 mhz

mcgurk, 2020-02-10:
I would first check antennas. Stock antennas can be for totally wrong frequencies and other uses than LoRa. Good antenna can be made from 86mm long single wire (for 868MHz). I get -25 rssi with 14dBm power (14dBm/25mW is maximum in my country).
[image]

IoTThinks, 2020-02-12:
With short and Chinese antenna, normally I only get around -50 to -40 RSSI within 20cm.

So it is normal in my viewpoint.
You can change to a better antenna but I heard the RSSI may not be so accurate within too-short distance.

beowulff, 2020-06-01:
I’ve never seen better than -50 or so at 2-3 feet distance, even with hand-tuned antennas that will transmit over 2 miles at ground level. So, I wouldn’t worry about trying to get much better than you currently have.
(I realize that this is an old topic, but other people might be interested.)

IoTThinks, 2020-06-01:
Really?
I intend to make my own antenna like @mcgurk to see if I can do better than -40 dBm RSSI.

beowulff, 2020-06-01:
Can you post a link to that Antenna?
I have a nice Anritsu spectrum analyzer, and I made a hand-tuned dipole connected to an SMA connector, and got excellent range, but still not better than around -48 at close range. The RF section on most of these LoRa modules is questionable - it’s hard to match an antenna, since there is no access to the antenna feed point before the matching network. 

Also, at close distances, you are within the Fresnel Zone, and tiny movements will have a big affect on performance, as will any object nearby.

IMHO, the best way to determine antenna performance is in the field - no obstructions and at least 100m of distance between transmitter and receiver.

IoTThinks, 2020-06-02:
You check the second post in this topic for the handmade antenna.

People said need to measure in at least several meters away.

Im just not happy enough for not reaching something better than -40 dBm RSSI.

May be the dipole or tripole antenna may help?

Your antenna is single wire antenna, right?

beowulff, 2020-06-02:
I made a  1/4 wave dipole antenna - two wires at 180° from each other.
I also have made a 1/4 wave monopole antenna, but that requires a large ground plane.

The antenna in post #2, is not optimal. It is a monopole, but has little to no ground plane. You can increase its efficiency greatly by adding an equal-length wire to the ground connection on the LoRa module, at 180° to the antenna wire.

IoTThinks, 2020-06-02:
Then I believe your test result is reliable.
Dipole antenna is often very good.

Just wonder how others get under -40dBm RSSI.

May be due to PA or RFO? Or other aspects?

beowulff, 2020-06-02:
Here is my Dipole antenna:
[image]

Note that Monopole antenna are slightly more efficient, but require a large (infinite is best) ground plane.

IoTThinks, 2020-06-06:
I use a 17+cm bronze wire as single wire antenna. 
Also get around 40+ dbm only.

One of my friends use the below antenna .
Simply soldered to Ai-Thinker ra01.
Can reach -24 or -15 RSSI within a few meters.

[image]

May be in short range, a good antenna can not give better RSSI number below -40dbm.
May be in long range, there will be different?

From RSSI and SNR, how to reliable test the quality of the signal?

closed 2020-11-12 by IoTThinks