sandeepmistry/arduino-LoRa #337: connecting two difrent nodes

kessero, 2020-02-13:
I am trying to communicate two devices. One broadcasting based on Ra-02 Aithinker SX1278 433MHz using arduino-LoRa library (demo sender). 
https://www.disk91.com/2019/technology/lora/first-steps-with-lora-radio-node-arduino/
Second receiving Waweshare LoRA_HAT SX1268 433MHz. Listening via USB UART. 
https://www.waveshare.com/wiki/SX1268_433M_LoRa_HAT#Registers_Configuration

There is silence in both sides even when I swap them. I don't quite understand some parameters. 

The receiver's (waveshare) channel is set to 23 because the current frequency is 410.125MHz + channel * 1MHz or 23 = 433.125 MHz. 
In arduino (sender) I set 433E6. Does this value after the decimal 125 matter?
I also set SignalBandwidth (62.5E3) and Air speed is 62.5K (receiver) to make it the same.
`void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender non-blocking");
  LoRa.setSignalBandwidth(62.5E3);
  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}`
Can these modules work together? What else do I have to set? 
Is the node ID set in the LoRa arduino library?
I would like to have two SX1278 senders for hat waveshare SX1268 on raspberry as gateway (no big trafic some temperature data).

Benedict-Bsl, 2020-02-13:
Sx1278 should basically work with its' pair. Sx1268 is a different module supported by its on kind. I have never tried those one out, but I am certain they can't communicate. You might perhaps need to do some tricks or adjustment on the libraries or so.

kessero, 2020-02-13:
So it's doesn't meter that all are 433MHz?

IoTThinks, 2020-02-14:
433.125Mhz is MUCH different from 433Mhz.
So 0.125Mhz is matter. 

You MUST have the same configuration on BOTH nodes.

Benedict-Bsl, 2020-02-14:
I am not sure if those modules could communicate they might have been build somehow differently, hence comes the difficulty (not that it's not possible) of pairing them together. Sx126x and Sx127x,... Try to pair two modules with same model number.

kessero, 2020-02-14:
> I am not sure if those modules could communicate they might have been build somehow differently, hence comes the difficulty (not that it's not possible) of pairing them together. Sx126x and Sx127x,... Try to pair two modules with same model number.

That's works well. I have to modules with SX1278 and i bought SX1276 as a hat for raspberrypi to collect data from 2x sx1278 and that's the problem.

morganrallen, 2020-02-14:
I cannot say for sure if the sx126x and sx127x series are compatible with each other, and Semtech doesn't say much I can find. But as @IoTThinks mentioned, it could be that 125kHz difference in frequency set point. It's not very well documented (if at all) but you can pass any frequency you want to `setFrequency`. Try calling `LoRa.setFrequency(433125000)`

IoTThinks, 2020-02-14:
I tried already.
Confirmed to pass LoRa.setFrequency(433125000). 433125000 is a "long".
It won't interfere to node with LoRa.setFrequency(433E6).

kessero, 2020-02-14:
I changed LoRa.setFrequency (433E6) to LoRa.setFrequency (433125000), but it still failed.
According to this page https://wiki.dragino.com/index.php?title=LoRa_Questions#How_many_LoRa_client_node_can_LG01.2FOLG01_support.3F
"If both devices have the same radio settings in LoRa: frequency, spread rate, bandwidth, preamble length and SyncWord. Both devices can communicate with anyone via LoRa. "
So I have to configure all parameters, but this hat has very poor access to configuration functions, so I have to try to configure it on SX1278 and this arduino library.

blacksmith289, 2022-06-01:
@kessero do you have any updates? Currently I have the similar situation as yours:
Using Waweshare LoRA_HAT SX1268 433MHz for RaspPi
And dragino's Arduino Wireless Shield with the library of https://github.com/mayeranalytics/pySX127x
Whould like to know what parameters you have modified
Many thanks!