sandeepmistry/arduino-LoRa #340: LORA-RA02 module

htarahi, 2020-02-22:
Hi every one
I have 2 arduino with 2 lora-ra02 modules that works in 433 MHz frequency.
I upload the code but it do not receive any thing !
both sides send packet but receive nothing !
what can be the problem ?
ofcourse some times sending stop working with no error.
[image]

IoTThinks, 2020-02-22:
Sending means nothing.
Only receiving counts the success.

Check the LoRa pin mapping, SetPins() and LoRa settings on both side.

htarahi, 2020-02-22:
This time I use the simplest code ...Lora sender & lora Receiver
again I have no receive !
I use arduino uno with lora 02 in both sides.
arduino board's Power is from USB, connected to PC.
arduino provides lora module power.
frequency is 433 (433E6) with pins connection as below.
[image]

htarahi, 2020-03-02:
no idea yet ??!

04-Hax, 2021-05-29:
Same Issue. Did you find the solution?

Kongduino, 2021-05-29:
Do you have an SDR ? It would allow you to verify that the packets are actually sent, and sent properly. I had an issue with a device once – the SDR showed it was sending, but at BW 250 KHz instead of the requested 125 KHz, or something similar. Which explained why other devices couldn't receive...

04-Hax, 2021-05-29:
Yes I have already cross checked it on Freq 433Mhz , and at the receiver side also it's tuned at 433Mhz

04-Hax, 2021-05-29:
I'm having Rx only , i check that LoRa is sending packets on frequency 433MHz but receiver side LoRa is not capturing anything.

IoTThinks, 2021-05-30:
Show your code, your wiring and real photo of your boards.

04-Hax, 2021-05-30:
I'm using example code , with 2 Arduino Uno r3 attached with Ai Thinker Ra-02.
[image]

IoTThinks, 2021-05-30:
@04-Hax With your own eyes on your board and your code, you can tell what is wrong.
How could anyone tell without seeing the code and real photo?

Wow, example codes. Which example codes? :D 
Must be a fortune teller to know.

Nevermind, let other people to help.

04-Hax, 2021-05-30:
Receiver Example.
At first it was giving me error , LoRa starting failed.Then I fixed it by making changes in cpp file.
Now everything is working good but the packets which are transmitted by LoRa is not getting received by another module.