sandeepmistry/arduino-LoRa #342: nrf52-lora(RA-02 module)

safwan9633, 2020-02-25:
Are you receiving `Starting LoRa failed` while using the demo code?

PLEASE see the [FAQ #1](https://github.com/sandeepmistry/arduino-LoRa#faq) about using [setPins](https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins) **BEFORE** submitting an issue.

hi 

i am using LoRa module RA-02 in  nrf52832  try to send 400bytes per sec but i can only send 256bytes per sec .how can i send 2 packet per sec(512bytes per sec)?

how can i increase datarates in lora without missing packets and bytes?

IoTThinks, 2020-02-25:
What are your current SF and BW (Hz) setting?

safwan9633, 2020-02-26:
hi
@IoTThinks  SF= 8,BW=125KHZ

IoTThinks, 2020-02-26:
SF6, BW500 is the fastest at 37kbps.
The faster the data rate, the shorter the distance.

safwan9633, 2020-02-26:
hi
@IoTThinks  i need 3kbps  with getting long distance
is it possible to send 2 packet(256bytes) per sec with sf 8 and bw 125khz?

IoTThinks, 2020-02-26:
Hi,
You can help yourself here
https://www.rfwireless-world.com/calculators/LoRa-Data-Rate-Calculator.html

closed 2020-11-12 by IoTThinks