sandeepmistry/arduino-LoRa #345: Is setTxPower doing what it's supposed to?

morganrallen, 2020-03-05:
I'm poking through some semi-related issues and I noticed something odd in `setTxPower` and would like to get some eyes on it. In this function, the values `0x87` and `0x84` are being written to `REG_PA_DAC (0x4D)` which doesn't seem to make sense. According to the datasheet this value only takes `0x04` or `0x07`. Though this value does make more sense for `REG_PA_CONFIG`.

I'm thinking these should be `0x07` and `0x04` unless I'm missing something.

IoTThinks, 2020-03-08:
Which pages of data sheet you are looking at?

I tried with txPower in this library. And only see up to 10dBm increase.

May be I'm over expected or something is not right.

Boromatic, 2020-08-19:
Seems in SetTxPower() the bit for PA_BOOST is set even if TX power should be below 17dBm (line 448). So there is only small dynamic of about 10dB.

closed 2020-11-12 by IoTThinks