sandeepmistry/arduino-LoRa #347: running on raspberry pi

loraraspi91, 2020-03-10:
Dear All,

For my personal needs, I port this arduino-Lora library code on raspberry pi, you can find all the code here:

https://github.com/loraraspi91/LoRa4Raspi

I also did the effort to port almost examples to raspberry pi. I kept the same API, so it is very easy to run an actual Arduino-Lora code on raspberry pi.

you are free to merge my port in this library, so this library could also supported by Raspberry pi.

Sorry, if it is not the right place to put this information.

Laurent

morganrallen, 2020-03-10:
Neat! I'll have to dig around and see if I have a LoRa on a module to test this.

IoTThinks, 2020-03-10:
@loraraspi91 Hi, It would be great if you have a basic sample wiring the LoRa sx1278 and a Pi.

Maybe a photo is sufficient. 
Thanks for the porting

loraraspi91, 2020-03-10:
I do not use specific wiring, I'm using the same pin like the original Lora
library

You just have to connect your sx1278 to SPI.
connect the NSS (Chip select) to a GPIO,
Reset to another GPIO
DIO0 to another GPIO
of course connect power of the Sx1278 to the Raspberry Pi Connector

after you use the LoRa.setPins( Nss, Reset, DIO0); you have to use WiringPi
number to specify it in the command.

For my test, I use CE0 pin for NSS of the raspberry pi (wiringpi =10); I
also use GPIO25 for Reset (WiringPi = 6) and I used GPIO19 (WiringPi 24)

but you can adapt it according to your board.

to be precise for your sx1278

1278  -->  Raspi connector
pin 8, DIO0 ---> pin 35, GPIO19, WPi 24
pin 17, MISO --> pin 21, MISO
pin 18, MOSI --> pin 19, MOSI
pin 16, SCK --> pin 23, SCLK
pin 19, NSS --> pin 24, CEO/GPIO8, WPi 10
pin 7, NRESET --> pin 22, GPIO25, WPi 6

LoRa.setPins (10, 6, 24);

IoTThinks, 2020-03-11:
@loraraspi91 Thanks for the detail.
It is sufficient for me already.

kizu29038988, 2020-11-30:
Thank you so much, could you make an comunication between Pi and Arduino (LORA SX1278, RA02) by TEXT. Thank you.

loraraspi91, 2020-12-05:
Hello,

I think it should work. I do not have the SX1278, RA02 Arduino module to
make the test. You have to be sure that the frequency and LoRa
configuration are the same on both side.

You can try with the sender.cpp example ( LoRa.print("hello ") ).

Best regards

closed 2021-02-19 by IoTThinks

Mathrim681, 2024-11-07:
If I try to compile the sender example with the comand line

$ g++ -Wall -o LoRaSender LoRaSender.cpp LoRa.cpp Print.cpp WString.cpp itoa.cpp -lwiringPi

I get->
command not found

if I try

$g++ -Wall -o LoRaSender LoRaSender.cpp LoRa.cpp Print.cpp WString.cpp itoa.cpp -lwiringPi
I get->
root@Database:/etc/GA_MK23/test/LoRa4Raspi/examples/LoRaSender# g++ -Wall -o LoRaSender LoRaSender.cpp LoRa.cpp Print.cpp WString.cpp itoa.cpp -lwiringPi
cc1plus: fatal error: LoRa.cpp: No such file or directory
compilation terminated.
cc1plus: fatal error: Print.cpp: No such file or directory
compilation terminated.
cc1plus: fatal error: WString.cpp: No such file or directory
compilation terminated.
cc1plus: fatal error: itoa.cpp: No such file or directory
compilation terminated.

if I copy the src files LoRa.cpp Print.cpp WString.cpp itoa.cpp in the folder of the LoRaSender
I get->

root@Database:/etc/GA_MK23/test/LoRa4Raspi/examples/LoRaSender# g++ -Wall -o LoRaSender LoRaSender.cpp LoRa.cpp Print.cpp WString.cpp itoa.cpp -lwiringPi
/tmp/ccX1G32o.s: Assembler messages:
/tmp/ccX1G32o.s:244: Error: unknown or missing system register name at operand 2 -- mrs x0,CPSR' /tmp/ccX1G32o.s:279: Error: unknown mnemonic cpsie' -- cpsie i' /tmp/ccX1G32o.s:312: Error: unknown mnemonic cpsid' -- `cpsid i'
WString.cpp: In member function ‘String& String::operator=(String&&)’:
WString.cpp:175:37: warning: ignoring return value of ‘constexpr typename std::remove_reference<_Tp>::type&& std::move(_Tp&&) [with _Tp = String&; typename remove_reference<_Tp>::type = String]’, declared with attribute ‘nodiscard’ [-Wunused-result]
175 | if (this != &rval) std::move(rval);
| ~~~~~~~~~^~~~~~
In file included from /usr/include/c++/12/bits/exception_ptr.h:43,
from /usr/include/c++/12/exception:168,
from /usr/include/c++/12/ios:39,
from /usr/include/c++/12/ostream:38,
from /usr/include/c++/12/iostream:39,
from /usr/include/WString.h:26,
from WString.cpp:19:
/usr/include/c++/12/bits/move.h:104:5: note: declared here
104 | move(_Tp&& __t) noexcept
| ^~~~
WString.cpp: In member function ‘String& String::operator=(StringSumHelper&&)’:
WString.cpp:181:37: warning: ignoring return value of ‘constexpr typename std::remove_reference<_Tp>::type&& std::move(_Tp&&) [with _Tp = StringSumHelper&; typename remove_reference<_Tp>::type = StringSumHelper]’, declared with attribute ‘nodiscard’ [-Wunused-result]
181 | if (this != &rval) std::move(rval);
| ~~~~~~~~~^~~~~~
/usr/include/c++/12/bits/move.h:104:5: note: declared here
104 | move(_Tp&& __t) noexcept
| ^~~~

takamasanumuro, 2024-11-07:
Username checks out

Mathrim681, 2024-11-07:
?

morganrallen, 2024-11-08:
This library does not work on Raspberry Pi, or any Linux for that matter. The link provided in the first comment is probably what you want.